MarlinFirmware/Marlin #19898: [BUG] Trying to use Servo0 for Piezo

GrevanGER, 2020-10-25:
Greetings. 
I am sitting here with my Marlin 2.0.x on an Anycubi i3 Mega S. Trying to get a precision piezo running. 
The piezo is running and setup, I have checked the wiring with my multimeter, everything seems to be fine. 
Yet the signals from the piezo are not effecting the Z axis in its movement when probing. 

The Trigorilla board on my printer is the TRIGORILLA_14 (No special markings on my board). 
According to the documentation, Servo0 should be pin 11 (or pin 5 when using TRIGORILLA_14_11).
So these are the changes I have made, tring to make things work.

#define ENDSTOPPULLUP_ZMIN_PROBE
#define Z_MIN_PROBE_ENDSTOP_INVERTING true
#define ENDSTOPPULLUP_ZMIN_PROBE
#define NOZZLE_AS_PROBE
#define AUTO_BED_LEVELING_BILINEAR
#define RESTORE_LEVELING_AFTER_G28

What am I missing?
//#define Z_PROBE_SERVO_NR 0 doesn't seem to be related to a probe without and actual servo on it. Or is it?

thisiskeithb, 2020-10-25:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-25 by thisiskeithb

GrevanGER, 2020-10-27:
I am not sure if this is actually "just a support request", since I have done everything in my power 
to make a servo pin work as signal input, but it is working. 

I think there is something wrong with the way the Trigorilla board is being handled after all, since someone decided to go "Oh yeah, pretty similar to a ramps board, lets take a shortcut and call it a ramps with special pins." At least that is my impression so far.