MarlinFirmware/Marlin #19903: [BUG] Bed MinTemp error triggered - even though bed thermal protection disabled

bipsendk, 2020-10-25:
### Bug Description

On my MKS Gen v1.4, I have suddenly recently seen a bunch of halts with Bed Mintemp errors - even though Bed thermal protection is disabled.
This happens on 2.0.5.3 - but the code suspected code looks the same in the 2.,0-bugfix branch.

### My Configurations

Not sure if they are relevant  - can be provided if needed

### Steps to Reproduce

Compile Marlin without bed thermal protection and install on controller.
Start print.
Monitor with Pronterface
At some point the print might fail with Bed Mintemp error - reason yet unknown

**Expected behavior:** [What you expect to happen]

Print would finish without errors

**Actual behavior:** [What actually happens]

Print fails with Bed MinTemp error 

#### Additional Information

I suspect the Bed MinTemp error to be triggered by this line in src/module/temperature.cpp:  

if (bed_on && BEDCMP(mintemp_raw_BED, temp_bed.raw)) min_temp_error(H_BED);

It is not enclosed by #ifdef (THERMAL_PROTECTION_BED )   / #endif

Monitoring the temperature with Pronterface just states:

RECV: ok T:214.43 /215.00 B:75.06 /75.00 @:90 B@:65
SENT: M105
RECV: ok T:213.13 /215.00 B:74.92 /75.00 @:127 B@:91
SENT: M105
RECV: ok T:214.95 /215.00 B:75.22 /75.00 @:75 B@:34
SENT: M105
RECV: ok T:215.00 /215.00 B:74.87 /75.00 @:76 B@:102
SENT: M105
RECV: ok T:215.47 /215.00 B:75.01 /75.00 @:66 B@:76
SENT: M105
RECV: ok T:214.58 /215.00 B:75.08 /75.00 @:89 B@:62
SENT: M105
RECV: ok T:213.70 /215.00 B:74.99 /75.00 @:111 B@:78
RECV: X:104.92 Y:119.18 Z:6.00 E:0.00 Count X:10647 Y:12080 Z:9600
SENT: M105
RECV: ok T:213.65 /215.00 B:74.97 /75.00 @:112 B@:81
SENT: M105
RECV: ok T:214.79 /215.00 B:75.02 /75.00 @:83 B@:71
SENT: M105
RECV: ok T:215.57 /215.00 B:75.02 /75.00 @:69 B@:71
RECV: X:118.52 Y:103.76 Z:6.00 E:0.00 Count X:10121 Y:7906 Z:9600
SENT: M105
RECV: ok T:216.51 /215.00 B:74.94 /75.00 @:43 B@:85
SENT: M105
RECV: ok T:214.17 /215.00 B:74.99 /75.00 @:103 B@:75
RECV: X:103.80 Y:76.15 Z:6.00 E:0.00 Count X:13257 Y:7037 Z:9600
SENT: M105
RECV: ok T:214.17 /215.00 B:75.00 /75.00 @:95 B@:74
SENT: M105
RECV: ok T:213.70 /215.00 B:74.83 /75.00 @:106 B@:104
SENT: M105
RECV: ok T:214.27 /215.00 B:74.99 /75.00 @:94 B@:74
RECV: Error:MINTEMP triggered, system stopped! Heater_ID: bed
[ERROR] Error:MINTEMP triggered, system stopped! Heater_ID: bed
SENT: M105
RECV: Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

The question is why the MInTemp is triggerd - if it somehow could be a single faulty reading/conversion of the AD converter where the thermistor is attached  ?

sjasonsmith, 2020-10-25:
This is most likely faulty wiring. If you slowly move your bed back and forth without printing, you may be able to find a location where your reading goes away.

This happened on one of my printer. The cable looked fine, but it was broken in several places inside the insulation.

ellensp, 2020-10-26:
Min temp and max temp is not part of thermal protection. Thermal protection looks at power being applied and temperature rise. If you applying power and temp in not increasing you trigger thermal protection. 
Min and Max are when the thermistor reading exceed your temperature tables. This will always halt your machine as their is a hardware fault.

closed 2020-10-26 by ellensp