MarlinFirmware/Marlin #19911: [BUG] "Change filament" from menu will always use second extruder on mixing extruder setup

fishcu, 2020-10-26:
### Bug Description

"Change filament" from menu will always use second extruder on mixing extruder setup.

This means that choosing "E1" will then change filament on the second extruder.
Choosing "E2" will change filament on the second extruder as well (as expected).

This is a regression. It was working fine before on a branch based on a much older commit:
https://github.com/fishcu/Marlin/tree/find-problem

### My Configurations

My config is here: https://github.com/fishcu/Marlin/tree/2.0.7.2-a10m

Flashed on a Geeetech A10M with GT2560 v3.0 board.

### Steps to Reproduce

As described above, use the menu to try and change filament on the first extruder.

**Expected behavior:** Change filament on the first extruder if I choose E1 in the menu.

**Actual behavior:** Filament is changed on E2 in any case.

#### Additional Information

It could be possible to do a `git bisect` between
bad: https://github.com/fishcu/Marlin/tree/2.0.7.2-a10m
good: https://github.com/fishcu/Marlin/tree/find-problem

fishcu, 2020-10-26:
Possibly related issue: https://github.com/MarlinFirmware/Marlin/issues/13351

fishcu, 2020-10-26:
I can confirm that sending `M600 T0` works as expected. I'm now looking into if maybe the command generated internally is somehow wrong.

fishcu, 2020-10-26:
Submitted a PR that fixes this here: https://github.com/MarlinFirmware/Marlin/pull/19915

closed 2020-10-29 by thinkyhead

nvibd, 2020-10-29:
This issue is not fixed. Why close it?

fishcu, 2020-10-29:
This issue is not fixed. Why close it?