MarlinFirmware/Marlin #19913: [BUG] (LEVEL_BED_CORNERS nozzle drags)

calini, 2020-10-26:
# Bug Description

Hitting 'Level Corners' in the Motion menu works okay if the steppers are already enabled, but if they have been disabled (either automatically or by hitting Disable Steppers) does not enable the Z stepper, therefore the nozzle drags along and scratches the bed surface.

### My Configurations

Running Marlin `bugfix-2.0.x`, fresh as of today.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5440755/config.zip)

### Steps to Reproduce

1. Move the nozzle away from the bottom left corner, so there is space for this to happen
2. Hit `Disable Steppers`, or wait for them to get disabled by timeout
3. Go to `Motion` -> `Level Corners`

**Expected behavior:** [What you expect to happen]

Z axis to rise 10mm, auto home, move to the bottom left corner, lower Z. 

**Actual behavior:** [What actually happens]

Z does not rise as X/Y move towards the bottom left corner, scratching the bed.

sjasonsmith, 2020-10-28:
It seems this needs to have some “unknown Z” logic like homing does.

When you have this problem, is your nozzle already resting on the bed, or does it fall as the X/Y move is performed?

qwewer0, 2020-10-28:
https://github.com/MarlinFirmware/Marlin/issues/19428 Related?

yysh12, 2020-10-30:
Yeah #19428 appears to be the same issue. Basically when steppers timeout automatically versus manually changes the default behavior for measures aimed at preventing bed scratches. The unknown Z logic fails on automatic timeout.

swissnorp, 2020-11-30:
See https://github.com/MarlinFirmware/Marlin/issues/19428#issuecomment-735906570 for possible solution please.

qwewer0, 2021-01-02:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/19428. Can be closed?

closed 2021-02-11 by github-actions[bot]