MarlinFirmware/Marlin #19914: [BUG] (gcode preview, mksui)

Malderin, 2020-10-26:
### Bug Description

In the generated gcode through the plugin "mks wifi", the model preview on the memory card is not displayed (displayed "no data")
Software - Cura v. 4.7.1, mks wifi plugin v. 1.0.2 (from cura store).

### My Configurations

[Conf.zip](https://github.com/MarlinFirmware/Marlin/files/5440936/Conf.zip)

### Steps to Reproduce

1. Install Cura slicer, install mks wifi plugin from cura store;
2. Slice any model and save on sd card with option "save as tft file";
[image]
3. Insert sd card into printer and click "printing"
4. "No data"
[image]

#### Additional Information

1. My test gcode file with the preview of the model:
[simple cube.zip](https://github.com/MarlinFirmware/Marlin/files/5440987/simple.cube.zip)
2. The preview is displayed normally during printing
[image]

rhapsodyv, 2020-12-25:
Fixed here:  #20564, can you test it?

Malderin, 2020-12-25:
> Fixed here: #20564, can you test it?
already better, but still there is a reception
[image]
[image]

rhapsodyv, 2020-12-25:
I was testing with just one file. Now it's ok, I guess. Can you try again? #20564

Malderin, 2020-12-25:
> I was testing with just one file. Now it's ok, I guess. Can you try again? #20564

everything is fine 
[image]

X-Ryl669, 2020-12-28:
Please reopen, the fix in #20564 does not work for me.
It still crash the printer when listing the files if one of them has a preview. I'm generating the preview like it's displayed above (via MKS Cura plugin)

This is the resulting file (unzip to a SD Card and try to list the files):
[SP_MU~1.GCO.zip](https://github.com/MarlinFirmware/Marlin/files/5747297/SP_MU.1.GCO.zip)
I can provide other if you want.

I'm using a MKS Robin Nano 35 (so it's a 480x320 screen and the expected icons are 100x100 not 50x50 as show above).

I've already fixed this in the MKS's [PR31](https://github.com/makerbase-mks/Marlin-2.x.x/pull/31). 
The fix is more complex than the proposed solution since there is both a bug in the file parsing and in the expected icon size.

Currently, the status of the MKS's UI is not clear. I can't really report bug on the PR#20191 since it's not in master, and if I'm reporting bugs to MKS's branch, it's delaying inclusion in Marlin as it has not settled yet.

Malderin, 2020-12-28:
@X-Ryl669 reopened

X-Ryl669, 2020-12-28:
With this fix, the printer crashes on the 3rd file (none displayed):
[image]

With my fix, the printer is happy:
[image]

(Sorry for the poor quality of the picture, I'm using the printer's webcam here)

rhapsodyv, 2020-12-28:
The file you attached is a invalid gcode. It have dirty on top, like part of a email. Just remove the dirty and it should work:
Dirty:
```
--boundary_.oOo._F2003otnPwbPRz13/F4h0s+GUovex2zD
Content-Disposition: form-data; name="file"; filename="SP_Mutter.gcode"

```

[image]

closed 2020-12-28 by rhapsodyv

X-Ryl669, 2020-12-28:
Sorry to bother you, but I haven't done anything to the file (it's what's produced by MKS WIFI Cura plugin). I have tens of such files on the SD card and they are all like this.

**AND** It doesn't matter since the code is correctly finding the first position of the `simage` part and storing this position in variable `sd_read_base_addr` as expected. It works by a matter of chance, since the code is searching for '\r' before a `simage` and there is one. 

Then,  remove them, and you'll see the file are **still crashing** the printer (because of the wrong size).
Please notice that GCODE files are text file and the end of line characters should not have any influence. Yet, the current code **expect** that lines are separated by '\r' char and not `\r\n` or `\n`. This means that opening a GCode file and saving them in any text editor (or any GCode processor) will mangle the end of lines and break the code expecting a `\r` only leading to a printer crash.

The bug is still present, removing the lines does not change anything. Please reopen.

rhapsodyv, 2020-12-28:
This should be enough: https://github.com/MarlinFirmware/Marlin/pull/20589

PR #20564 (2020-12-28): Fix G Code Preview for LVGL_IU
### Description

Fix #19914 

It was broken by recent changes on MKS Cura plugin and the watchdog enabled by default.