MarlinFirmware/Marlin #19917: MKS SGEN_L V2 unable to access SD card from PC via USB

mks-viva, 2020-10-27:
### Bug Description

- Use MKS SGEN_L V2 motherboard and use the latest version bug fix marlin
  - Enable  #define SDSUPPORT unable to access SD card from PC via USB
  - Disable #define SDSUPPORT, can be access SD card from PC via USB
  - Issues: https://github.com/makerbase-mks/MKS-SGEN_L-V2/issues/25

### My Configurations
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5442348/config.zip)

ellensp, 2020-10-27:
The sdcard is either accessed by marlin or the computer, It cannot be both.
If marlin has control, you need to M22 - Release SD card (or use lcd)
If the OS has control you need to M21 - Init SD card  (or use lcd)

ellensp, 2020-10-27:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

rhapsodyv, 2020-10-27:
I can confirm this issue. Neither SD attach or released shows anything in the computer.

Even worst. I just found that I cannot boot the board if I have a SD in and the USB connected to the computer.

ellensp, 2020-10-27:
there may be something to this...

rhapsodyv, 2020-10-28:
@mks-viva can you test this: #19932? It's more like a temporary fix, but may work.

mks-viva, 2020-10-28:
@rhapsodyv 
I have tested it, but it has not improved. But i find it can be access SD card from PC via USB after Release Media.

rhapsodyv, 2020-10-28:
Yes, release media is needed to access the SD from the computer. 

Tomorrow I will do more tests too.

rhapsodyv, 2020-11-28:
Go away bot. @p3p can you merge my PR so I can fix this?

RP-projekts, 2020-11-28:
Hi there.. I tried out your changes  - result getting a new error after releasing the SD-Card 
before it was "no partition - media is not formated" --- now i get "please insert media"
isn't it possible making a new switch like "release SD to be used as massstorage" and set an watchdog on the powerline to automatically reset this option on power down

RP-projekts, 2020-11-28:
@mks_viva thanks --- this is exact what i'm looking for -- even it doesn't resolve my problems by now

RP-projekts, 2020-11-28:
@rhapsodyv hello and thank's for your work ---  i found out, that on the stm32f1, they are dealing with the same problem -- maybe this can help you for fixing the problem with LPC

rhapsodyv, 2020-11-28:
@RP-projekts what is your board and configuration? I can take a look if you attach here.

RP-projekts, 2020-11-28:
Hi -- I use an MKS SGEN_L V2.0 (LPC1769) with the MKS TS35 Touchscreen , ESP8266 Wifi Modul via Serial_Port2 (with ESP3D lastversion) .. cause of TS35 instead of buying an TFT35 i've to use everything onboard (SDCard and Wifi)

I'm exact at the same point like moham96 in #19939
my config and config_adv are in the zip
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5611069/Marlin.zip)

RP-projekts, 2020-12-06:
@rhapsodyv Hello Victor, thanks to your help, I managed to use the board successfully. But I have one more question? Is there no way to find a solution where I don't have to plug in / unplug the USB after releasing the SD?

wish you and your family a nice Advent and thank you again for your good work

thisiskeithb, 2024-06-02:
Reopened since https://github.com/MarlinFirmware/Marlin/pull/19939 was closed & not merged.

thinkyhead, 2026-10-08:
This looks to have been addressed by #19932, merged since this report, and there's been no further activity here. Closing for now as part of 2.1.3 cleanup. If you still see the problem on current `bugfix-2.1.x`, please let us know and we'll reopen.

PR #19932 (2020-10-29): Never Stop SPI on LPC
### Description

LPC MSC code start and configure the SPI 1 when the SD is release (to be shared with the host). After that initialisation, it relies that the SPI 1 will be always enabled and with the right configuration.

It's work fine with boards that keep SPI 1 only for SD.

But, some boards, like MKS SGEN L V2, share SPI 1 for the SD and the LCD/TFT. This makes SPI 1 changes constantly.

This results in random things, like not being able to mount SD, or even shows the SD as corrupted in the HOST.

This is just a temporary (but OK) fix, until we check if the LPC frame [... cut]
closed 2026-10-08 by thinkyhead