MarlinFirmware/Marlin #19920: [BUG] Compilation failure in dev mode for some platforms

0r31, 2020-10-27:
Hi all,

### Bug Description

When enabling **MARLIN_DEV_MODE**, new Dn codes are loaded from Marlin/src/gcode/gcode_d.cpp.
In case persistent storage is used (**EEPROM_SETTINGS** enabled), the eeprom size is defined using **E2END** (check [this line](https://github.com/MarlinFirmware/Marlin/blob/8cf672104efb8f7093c2e85a00b5d98993d368fc/Marlin/src/gcode/gcode_d.cpp#L95)).
Unfortunately, **E2END** is not defined for most platforms which leads to compilation failure.

It is confirmed by the way the eeprom size is defined (mostly hardcoded to 4K) in all HAL eeprom stuff.

### Work in progress

I am trying to find a smart way to fix this issue through a pull request.
Any advice is welcomed.

Regards,
Orel

shitcreek, 2020-10-29:
Perhaps something like this:
`#define MARLIN_EEPROM_SIZE size_t(TERN(__AVR__, E2END + 1, 0x1000));`

0r31, 2020-10-29:
Hi @shitcreek ,

Yep that's the quick win fix. 
One can also test if E2END is defined as it is used on other platforms:
```c
TERN(E2END, E2END + 1, 0x1000)
```

Unfortunately, MARLIN_EEPROM_SIZE is defaulted to other values in HAL eeprom stuff:
- In [Marlin\src\HAL\LPC1768\eeprom_wired.cpp](https://github.com/MarlinFirmware/Marlin/blob/ee7476a0bfcfd9c894fe99398f456a4fe2426209/Marlin/src/HAL/LPC1768/eeprom_wired.cpp#L37)
- In [Marlin\src\HAL\STM32F1\eeprom_flash.cpp](https://github.com/MarlinFirmware/Marlin/blob/ee7476a0bfcfd9c894fe99398f456a4fe2426209/Marlin/src/HAL/STM32F1/eeprom_flash.cpp#L43)

I think one must reflect those default values. If so, testing the target or E2END definition is not enough as it also depends on eeprom type.

Regards,
Orel

shitcreek, 2020-10-29:
We could simply do:
```
#ifndef MARLIN_EEPROM_SIZE
  #error "MARLIN_EEPROM_SIZE is required for EEPROM."
#endif
```
And add it to our Sanity check

0r31, 2020-10-29:
I guess the right thing to do is to use PersistentStore implementation.

It leads to replace
```c
#ifndef MARLIN_EEPROM_SIZE
   #define MARLIN_EEPROM_SIZE size_t(E2END + 1)
#endif
NOMORE(addr, (size_t)(MARLIN_EEPROM_SIZE - 1));
NOMORE(len, MARLIN_EEPROM_SIZE - addr);
```
by
```c
NOMORE(addr, (size_t)(persistentStore.capacity() - 1));
NOMORE(len, persistentStore.capacity() - addr);
```

commit f8d57370 (2020-10-29): gcode_D EEPROM size, cleanup

Fixes #19920