MarlinFirmware/Marlin #19923: [FR] Color UI - mount SD card after touching SD icon on status screen

bkiziuk, 2020-10-27:
# Description

The current behavior when SD card is not mounted (for example, after Power On without SD card inserted) is that the SD card icon is greyed out.
So, when the SD card is inserted, in order to mount it, it is required to go to Settings, scroll down to the bottom and choose Attach Media.
So the FR is simply to try to attach media, when the SD icon is touched and SD card is unmounted.

### Feature Workflow

1. Power-on printer without SD card inserted
2. Insert SD card to the slot
3. Touch greyed out SD icon in order to mount the SD card and then display its contents

#### Additional Information

None.

rhapsodyv, 2020-10-27:
It’s a good idea... 

But the meaning of the gray button is: this button is disabled. 

Making a disabled button touchable, may break the design language of the interface.

But maybe if we use different color or icon for status, could work.

I will check this.

Diego-Rocha, 2020-10-30:
@rhapsodyv maybe use the same color/state as enabled...
and if clicked and fail to mount, just show the message on status "SD Init fail", or something like that.
let me know what do you think?

rhapsodyv, 2020-11-02:
I just made the button always enabled and entering the MenuMenu. In the menu media, the refresh button will mount the SD if it is unmounted! This way we don't break the design language and support what you have suggested! :-)

I hope you like: #20005

Thanks!

bkiziuk, 2020-11-03:
I think I can live with this solution. Much better than going through the menu :)

PR #20005 (2020-11-03): MenuMedia always accessible from Color UI status screen
### Description

MenuMedia have a option to mount the SD if it's unmounted (clicking in refresh). So we don't need to disable the access to this menu from status screen.

This PR make the MenuMedia always accessible from the SD shortcut in the color ui status screen.

### Benefits

Easy to mount SD.
Fix: #19923
closed 2020-11-03 by bkiziuk