MarlinFirmware/Marlin #19924: using FAST_PWM_FAN yelds only 0% and 100% duty cycle

f-dumi, 2020-10-27:
Hello,

I have the same issue as in #16115 but there is no solution stated there... 

I am using the latest `bugfix-2.0.x` build on an Ender 3 Pro with a standard Creality3D V1.1.4 board.
I am using some Sunon fans for print cooling, which appear to have a fast-acting undervoltage lockout, and therefore the ~976Hz PWM used to drive the fans is not fast enough (fans spinning at only >98% speed). I tried enabling `FAST_PWM_FAN` but now there are only 2 levels, 0% and 100% confirmed also with the original fans; also the frequency has not changed (confirmed with oscilloscope on OCR0B pin of the ATmega1284P)

I use the Ender 3 Pro (CrealityV1) configuration from github, to which I applied the below modifications.

## Steps to reproduce
1. uncomment `#define FAST_PWM_FAN` in Configuration.h
2. comment `#define SPEAKER` (#13571) in Configuration.h
3. uncomment '#define FAST_PWM_FAN_FREQUENCY 31400' and set whatever frequency above 20kHz
4. adjust fan speed via encoder to whatever value between 1% and 99%

## Expected behavior
Increase of the PWM frequency on the OCR0B output and smooth speed control via encoder

## Actual behavior
PWM frequency is the same as without modifications (976.6Hz).
Starting with speed  at 0% and setting speed to 100%, sets the speed to 100%, for other values between 1% and 99% the speed stays at 0%.
Starting with speed  at 100% and setting speed to 0%, sets the speed to 0%, for other values between 1% and 99% the speed stays at 100%.

ellensp, 2020-10-28:
So you using pin D4
Looking at set_pwm_duty, its being called correctly  
But one of the first checks is get_pwm_timer this returns 0. With a large comment beside it
// Don't proceed if protected timer or not recognised (ie 0) 
So yes this does exactly as you say it is doing.  full on or full off. Otherwise it just does nothing as I thinks It should not change this timer.

ellensp, 2020-10-28:
Timer 0 is used by the TEMPERATURE_INTERRUPT, so you cannot use this and pin4 with FAST_PWM_FAN.

sjasonsmith, 2020-10-30:
I recall having to use `FAN_SOFT_PWM` on my stock Ender 3.

sjasonsmith, 2020-10-30:
Well, not stock, with my new 5015 blower fan.

f-dumi, 2020-10-30:
I have also tried `FAN_SOFT_PWM`, but could not increase the frequency.
I also tried modifying all temperature related code and the Arduino base code to use Timer2, but there are still some instructions (that I couldn't find) that want to use Timer0 so I still could not make it work at high frequency.

I ended up modifying the stock Ender PCB to swap the OC0B output with the OC2A output (STEP X signal) and swapped the signals also in Marlin. This way I could set Timer2 for the fans and now it works fine.

Maybe it should be specified directly in `Configuration.h`  (at least for Ender 3) that `FAST_PWM_FAN` does not work if the fan(s) are controlled by the same timer as the one used in the temperature control algorithms.

closed 2020-10-30 by f-dumi