MarlinFirmware/Marlin #19927: Having trouble compiling the latest bugfix 2.0.x

NicoletoAerials, 2020-10-27:
### Bug Description

When trying to compile the latest bug fix version I'm getting the following errors. 

It is important to note that when compiling the main branch I get NO errors at all.

[image]

'C:\Users\Nicolas' is not recognized as an internal or external command,
operable program or batch file.
CalledProcessError: Command 'C:\Users\Nicolas Carrizosa\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-g++.exe -DTARGET_STM32F1 -D__MARLIN_FIRMWARE__ -DHAVE_SW_SERIAL -DDEBUG_LEVEL=0 -DSS_TIMER=4 -DSTM32_FLASH_SIZE=512 -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\Nicolas Carrizosa\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(item, exports="env")
  File "C:\Users\Nicolas Carrizosa\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Nicolas Carrizosa\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:     
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Nicolas Carrizosa\Documents\GitHub\Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 287:
    apply_features_config()
  File "C:\Users\Nicolas Carrizosa\Documents\GitHub\Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 118:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\Nicolas Carrizosa\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Nicolas Carrizosa\Documents\GitHub\Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 263:
    load_marlin_features()
  File "C:\Users\Nicolas Carrizosa\Documents\GitHub\Marlin\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 251:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\Nicolas Carrizosa\.platformio\python3\lib\subprocess.py", line 411:
    **kwargs).stdout
  File "C:\Users\Nicolas Carrizosa\.platformio\python3\lib\subprocess.py", line 512:
    output=stdout, stderr=stderr)
=============================================================== [FAILED] Took 1.28 seconds =============================================================== 

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RC_btt_512K  FAILED    00:00:01.281
========================================================== 1 failed, 0 succeeded in 00:00:01.281 ==========================================================
The terminal process "C:\Users\Nicolas Carrizosa\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

Install fresh VS Studio Code in Win 10 64bits
Install Platformio platform
Compile the latest Bugfix version

**Expected behavior:**

**Actual behavior:**

#### Additional Information

Rukkulko, 2020-10-28:
Same happens to me it.
For some reason it breaks when there is a space in the path

ManuelMcLure, 2020-10-28:
Please provide your configuration files _as required by the bug template_.

rhapsodyv, 2020-10-28:
Check if this solves your issue: #19929

Rukkulko, 2020-10-28:
Your change has indeed resolved the issue in my case!

NicoletoAerials, 2020-10-28:
@rhapsodyv Great!!! It did work. Thanks!!

PR #19929 (2020-10-29): Quote compiler call to not break when compiler path have spaces
### Description

Dep script call uses the full path now. But some users have spaces in the path where the pio is installed. This quote the compiler call.

### Benefits

Fix #19927 

### Related Issues

#19881
closed 2020-10-28 by NicoletoAerials