MarlinFirmware/Marlin #19930: [FR] Compensation for noisy filament sensors

Liger0, 2020-10-28:
### Description

Hi. The issue I am actually having is that sometimes, on the cr10S, the filament sensors triggers Marlin to pause and directly unload filament, but when I check the filament is still available and the sensor sees it correctly.
I believe there is some "noise" with the sensor and it sees it as not triggered for only a moment.
I wonder if there is the possibility to make it trigger the pause/unload function only if the sensor is open for a longer interval instead of acting immediately.
Unfortunately I can't confirm that it effecively is detected as open in those moments, I am unsure how to verify it, so in case, instruct me about it please.

### Feature Workflow

1. Filament sensor detected open
2. After an interval, like let's say 0.5 seconds (configurable), check it again
3. Only proceed pausing the print and unloading the filament if the sensor is still detected open

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

yysh12, 2020-11-04:
Pretty sure this is already implemented in Marlin. I've set mine to a delay of 7mm before triggering filament change routine.

yysh12, 2020-11-04:
#define FILAMENT_RUNOUT_DISTANCE_MM 10

thisiskeithb, 2020-11-04:
> FILAMENT_RUNOUT_DISTANCE_MM

That just allows your to keep printing for a configured distance after runout is detected:
https://github.com/MarlinFirmware/Marlin/blob/a1cce36c1ed4d551ce699f13ba49d7fb3c2ce328/Marlin/Configuration.h#L1196-L1199

If your filament runout sensor is sending false positives, increase `FILAMENT_RUNOUT_THRESHOLD`:
https://github.com/MarlinFirmware/Marlin/blob/a1cce36c1ed4d551ce699f13ba49d7fb3c2ce328/Marlin/src/feature/runout.h#L45-L47

Closing since this is already implemented as noted above.

closed 2020-11-04 by thisiskeithb

Liger0, 2020-11-05:
> If your filament runout sensor is sending false positives, increase `FILAMENT_RUNOUT_THRESHOLD`:

Thank you for pointing this out.

Differently than the ENDSTOP_NOISE_THRESHOLD feature, this one doesn't seem documented, so I have no idea in which range to play, and how do I know if a value is too high.

yysh12, 2020-11-05:
Trial and error. Keep increasing until you no longer get false positives.

thisiskeithb, 2020-11-05:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/18861

Liger0, 2020-11-05:
Okay, but what's the order of magnitude to play it with? What is the deal if it is set too high?

thisiskeithb, 2020-11-05:
> Okay, but what's the order of magnitude to play it with? What is the deal if it is set too high?

I haven't used this setting, so one at a time? Too high would mean it would never trigger.

thisiskeithb, 2020-11-05:
For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y