MarlinFirmware/Marlin #19931: [BUG] Inductive probe always TRIGGERED

spprod35, 2020-10-28:
Hi 

i'am reopen this issue #6603 .
I have this interface on my ramps 1.4 : https://www.bbfil.fr/carte-d-interfacage-de-capteur-inductif-c2x22858635
With Marlin 1.0 my inductive 6-36v probe works perfectly.
With Marlin 1.1.9 my z_min or z_probe stay TRIGGERED always

Do you have an idea ?

thx for help and sorry for my english

thisiskeithb, 2020-10-28:
Marlin `1.1.x` is no longer being actively worked on. Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

Also, please attach configs like the Issue template asked for.

ellensp, 2020-10-28:
No Configs, hasn't follower template. Using V1 of marlin. Clearly a support question.   

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-28 by ellensp