MarlinFirmware/Marlin #19933: [BUG] (short description) MKS SGEN L V1.0 LPC1768 and MKS TS35 compilation error

dredivan, 2020-10-28:
Latest Marlin BUX FIX board **MKS SGEN L V1.0** LPC1768 
on clean configuration I edit only this parameters
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
#define MOTHERBOARD BOARD_MKS_SGEN_L
#define MKS_TS35_V2_0
#define TFT_COLOR_UI
--------------------------------------------------------------------------------------------------------------
error:
**Marlin\src\HAL\LPC1768\tft\tft_spi.h:63:50: error: 'TFT_A0_PIN' was not declared in this scope; did you mean 'TFT_SPI'?**

**Marlin\src\pins\lpc1768\pins_MKS_SGEN_L.h**  in this file no records with parameter TFT_A0_PIN

-----------------------------------------------------------------------------------------------------------------

https://github.com/makerbase-mks/MKS-SGEN_L-V2/blob/master/Firmware/Marlin-bugfix-2.0.x-MKS%20TS35%2BSGEN_L%20V2/Marlin-bugfix-2.0.x/Marlin/src/pins/lpc1768/pins_MKS_SGEN_L.h
here is  records
```
#if HAS_SPI_LCD
  #if ENABLED(CR10_STOCKDISPLAY)
    #define LCD_PINS_RS                    P1_00
    
    #define BEEPER_PIN                     P1_31
    #define BTN_ENC                        P1_30
    #define BTN_EN1                        P0_18
    #define BTN_EN2                        P0_15

    #define LCD_PINS_ENABLE                P1_22
    #define LCD_PINS_D4                    P0_17
  #elif HAS_SPI_TFT                               // Config for Classic UI (emulated DOGM) and Color UI
    #define TFT_CS_PIN                     P1_00
    #define TFT_A0_PIN                     P1_22
    #define TFT_DC_PIN                     P1_22
    #define TFT_MISO_PIN                   P0_08
    #define TFT_BACKLIGHT_PIN              P0_18
    #define TFT_RESET_PIN                  P0_16
```
-----------------------------------------------------------------------------------------------------------------------------
Here is log:
```

> Executing task in folder Marlin-bugfix-2.0.x_2020.10.28_Clear: C:\Users\USER\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
 - framework-arduino-lpc176x 0.2.5
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
Building in release mode

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1768/firmware.bin) manually to the appropriate disk

Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\DebugMonitor.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\HAL_SPI.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\MarlinSerial.cpp.o
Compiling .pio\build\LPC1768\src\src\HAL\LPC1768\eeprom_flash.cpp.o
[... 77 lines cut ...]
      |                                                                                             ^~~~~~~~~~
Marlin\src\HAL\LPC1768\tft\../../../inc/../HAL/./LPC1768/fastio.h:52:49: note: in definition of macro 'WRITE_PIN'
   52 | #define WRITE_PIN(IO,V)       LPC176x::gpio_set(IO, V)
      |                                                 ^~
Marlin\src\HAL\LPC1768\tft\../../../inc/../HAL/./LPC1768/fastio.h:105:35: note: in expansion of macro '_WRITE'
  105 | #define SET_OUTPUT(IO)        do{ _WRITE(IO, _READ(IO)); _SET_OUTPUT(IO); }while(0)
      |                                   ^~~~~~
Marlin\src\HAL\LPC1768\tft\../../../inc/../HAL/./LPC1768/fastio.h:115:35: note: in expansion of macro 'SET_OUTPUT'
  115 | #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)
      |                                   ^~~~~~~~~~
Marlin\src\HAL\LPC1768\tft\tft_spi.h:63:83: note: in expansion of macro 'OUT_WRITE'
   63 |   static void WriteReg(uint16_t Reg) { OUT_WRITE(TFT_A0_PIN, LOW); Transmit(Reg); OUT_WRITE(TFT_A0_PIN, HIGH); }
      |                                                                                   ^~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LPC1768\src\src\HAL\LPC1768\tft\tft_spi.cpp.o] Error 1
=========================================================================== [FAILED] Took 11.76 seconds ===========================================================================

Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:00:11.761
====================================================================== 1 failed, 0 succeeded in 00:00:11.761 ======================================================================
Процесс терминала "C:\Users\USER\.platformio\penv\Scripts\platformio.exe 'run'" был завершен с кодом выхода 1.

Терминал будет повторно использоваться задачами. Чтобы закрыть его, нажмите любую клавишу.
```

rhapsodyv, 2020-10-29:
@dredivan take a look if it works for you: https://github.com/MarlinFirmware/Marlin/pull/19945

dredivan, 2020-10-29:
> @dredivan take a look if it works for you: #19945

with new **pins_MKS_SGEN_L.h**
the display shows a color image, encoder is working, but everything works very slowly
the touchscreen does not work

on work config don`t want compilie with **#define LCD_BED_LEVELING**

rhapsodyv, 2020-10-29:
Did you try to calibrate the touch with M995?

But if it work too slow, maybe that board is not enough speed to run color ui.

Try the classic ui.

dredivan, 2020-10-29:
> Did you try to calibrate the touch with M995?
> 
> But if it work too slow, maybe that board is not enough speed to run color ui.
> 
> Try the classic ui.

https://github.com/makerbase-mks/MKS-SGEN_L-V2/tree/master/Firmware/Marlin-bugfix-2.0.x-MKS%20TS35%2BSGEN_L%20V2

I compile this config for MKS SGEN L v1 LPC1768, touch screen work. No need calibrate the touch. But after some time board reboots.

rhapsodyv, 2020-10-31:
Can you test again my branch?

sjasonsmith, 2020-11-12:
I verified that I can build with the original configuration changes specified. I'm presuming this is fixed.

closed 2020-11-12 by sjasonsmith