MarlinFirmware/Marlin #19935: Probing fails at last probe point

Ski3d, 2020-10-28:
I believe i found a bug?lol 
Anyways... if you disable 
//#define USE_ZMIN_PLUG
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
Enable
#define USE_PROBE_FOR_Z_HOMING
#define Z_MIN_PROBE_PIN 32 // Pin 32 is the RAMPS

On the very last probe point, it will fail probing everytime. 
But if i enable #define USE_ZMIN_PLUG, the probing will Not fail at the last probe point. And im not switching any wires on my GTR v1.0. And using the pin off the Bltouch port for my inductive probe. So no changes made other then just enabling the zmin plug.
[image]
[image]

Even if its not a bug, someone can use search option and find this to maybe fix there problem if it happens to them.
👍👍

ellensp, 2020-10-28:
Please attach your actual configuration files. So we can check version number for eg.

Ski3d, 2020-10-28:
Well i guess that was not the fix, did a 25 point level and on the last probe, as soon as the light on the Inductive sensor lights up it will fail 
Recv:  T:44.11 /0.00 B:35.40 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:43.88 /0.00 B:35.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error: Probing Failed"
Send: M112
Send: N3 M112*34
Send: N4 M104 T0 S0*37
Send: N5 M140 S0*96
Changing monitoring state from "Error: Probing Failed" to "Offline (Error: Probing Failed)"
Connection closed, closing down monitor

Ski3d, 2020-10-28:
[Desktop.zip](https://github.com/MarlinFirmware/Marlin/files/5455134/Desktop.zip)

Ski3d, 2020-10-29:
[Config update.zip](https://github.com/MarlinFirmware/Marlin/files/5455399/Config.update.zip)
Swapped to z min end stop pin , No difference. all my TMC2209 Pins been cut but the X and Y. Sensor less homing on XY and 4 Z independent stepper drivers

closed 2020-10-29 by Ski3d

shitcreek, 2020-10-30:
could you tell us how if you did resolve the issue?