MarlinFirmware/Marlin #19936: [BUG] G28 Home position

DrumClock, 2020-10-28:
Why is the home position set to **X = 115mm**  and  **Y = 90mm**,
when is HotBed 240x240 mm?

```
#define X_BED_SIZE 240
#define Y_BED_SIZE 240

#define Z_SAFE_HOMING_X_POINT X_CENTER
#define Z_SAFE_HOMING_Y_POINT Y_CENTER
```

GMagician, 2020-10-28:
I try to read the magic sphere...you have the Z_SAFE_HOMING option enabled

**In the future attach your configuration as required by template**

DrumClock, 2020-10-28:
Hi @GMagician 
Yes, I have the `Z_SAFE_HOMING` option enabled.

It's as I write above, it focuses on the Z-probe. If I set
`NOZZLE_TO_PROBE_OFFSET {0, 0, 0} `position after G28 is
X = 120 and Y = 120 mm.

Probably not a mistake.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5453381/Configuration.zip)

closed 2020-10-28 by DrumClock

GMagician, 2020-10-28:
> Probably not a mistake.

No, its purpose is to home Z in the center of the bed, then since home is done with probe, it has to be in the center, not the nozzle.