MarlinFirmware/Marlin #19937: [BUG] Z drift in ABL_Bilinear along the Y axis

LarrxX, 2020-10-28:
### Bug Description
G29 returns a grid with higher and higher Z values along the Y axis. The bigger the grid, the higher the Z values.

### Configuration Files
[ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/5453487/ConfigFiles.zip)

### Steps to Reproduce
Simply run a G29 command with a high number of grid points (9x9 for example).

**Expected behavior:**
No matter the size of the grid, get coherent Z-Values along the bed.

**Actual behavior:**
The larger the grid used for ABL, the higher the Z-Values get along the Y-axis.

#### Additional Information
I used the [latest commit](https://github.com/MarlinFirmware/Marlin/commit/12dec2563afcc0f92333ce3ddbba28979bc0f865) from the bugfix-2.0.x branch.

I enabled `DEBUG_LEVELING_FEATURE `and `M114_DETAIL`. I used different size grids (3x3, 5x5, 8x8, 11x11x 3x11, 11x3) and ran the following:
`M111 S247`
`G28`
`G29`

You can find the logs in the following zip file:
[Logs.zip](https://github.com/MarlinFirmware/Marlin/files/5453545/Logs.zip)

And here are the results as seen in the Octopi Bed Visualizer Plugin:
[image]

This could probably also be the cause for issue #18769 ?

OC-Rookier, 2020-10-28:
Interesting. The bed level wasn’t changed at any point during or in between these tests? I’ll run similar tests on my machine and post the results here.

LarrxX, 2020-10-28:
Nope, didn't touch the bed between tests

romejoe, 2020-10-28:
@LarrxX I tried to look at your logs.zip, but it contains your configuration files.

LarrxX, 2020-10-28:
Oops! Updated the log files zip in the original post, sorry!

Talha909, 2020-10-29:
any success

Bloodpack, 2020-10-30:
Here same Problem 

Z drift in ABL_Bilinear along the Y axis

Printer CORE XY

blueeagle69, 2020-11-01:
following!

romejoe, 2020-11-02:
@LarrxX I have been looking into this issue(unfortunately slowly because of time restrictions and I'm new to the Marlin project) and I had a couple questions. 

- What kind of bed are you using? The bed that comes with the Ender 3 Pro or a glass one?
- Your Ender 3 Pro is using the original 1.x Creality main board, correct?
- If you have a chance, can you rerun the 11x11 bed leveling with `G29 V4`?

I haven't had a chance to test this on my Ender 3 Pro(4.2.2 main board) yet. Hopefully I can run them either tonight or tomorrow.

LarrxX, 2020-11-02:
@romejoe answers to your questions:

- The original bed that came with the Ender 3 Pro (flexible magnetic bed)

- I bought the printer back in march and never changed the board, so yes, I assume it's using the original 1.x Creality main board.

- Here's my log running `G29 V4 [serial.log.txt](https://github.com/MarlinFirmware/Marlin/files/5476594/serial.log.txt) and the visualization result 
[image]

`

xyzz, 2020-11-02:
I've had a similar thing happen before, try running probe repeatability test in the middle of the bed (`M48 V4`), if you see Z constantly increasing, then the Z lead screw coupler needs to be tightened.

romejoe, 2020-11-03:
@LarrxX I wasn't able to reproduce this issue on my Ender 3 Pro :/ I'm not sure if this is related, but this printer is fairly new(only a couple weeks old and a handful of small prints). 

[image]

[grid.log](https://github.com/MarlinFirmware/Marlin/files/5479165/grid.log)

Here are the configuration files that I used:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5479193/Configuration.zip)

I have been busy tonight trying to reproduce on my print and haven't had a chance to look over those logs. Have you tried @xyzz 's solution?

LarrxX, 2020-11-03:
@xyzz was right. I did a 50 probe repeatability test `M48 P50 V4` and got a standard deviation of **0.115**!! After tightening the Z lead screw, the same test gives a deviation of **0.003** which is way way WAY better :)

And the 11x11 G29 after tightening the screw results in this:
[image]
(no my bed is not flat, which is why I got the BLTouch in the first place).

So I guess my issue is resolved and it's not a Marlin bug.

Thanks for the help!

(will wait a day if there are any answers before closing the issue)

OC-Rookier, 2020-11-03:
I just ran this test and got a standard deviation of about .006, which I think is 
[pretty good](https://github.com/MarlinFirmware/Marlin/files/5481835/M48.test.txt). I also have the bed visualizer plugin for Octoprint, which also shows good [image]. However, I do not think this is the full explanation for #18769.

LarrxX, 2020-11-03:
This actually did solve issue #18769  for me, was just heading there to comment on it.

romejoe, 2020-11-03:
I'm planning to look into #18769 next. I think I have been having that problem with my CR-10 S5. I'll post over there with any info I have. @LarrxX I'm glad that fixed your issue!

schleir, 2020-11-03:
I'll try the test today and post the results.

schleir, 2020-11-04:
My Z lead screw coupler is not loose.
I verified that my repeatability test `M48 P50 V4` has less than 0.004 Standard deviation on 4 corners + center.

I don't have this issue(Z drift), but still am observing issues reported in https://github.com/MarlinFirmware/Marlin/issues/18769

LarrxX, 2020-11-04:
I'm closing the issue since it's not Marlin related.

closed 2020-11-04 by LarrxX