MarlinFirmware/Marlin #19940: [BUG] (Delta bed leveling incorrect probe result  )

gellary, 2020-10-29:
### Bug Description
I have 2 problem 

1.  bed leveling concave occur 
2.  G29 odd result

I test with newest Marlin 2.0 Bugfix and marlin 2.0.7 the concave and strange G29 result still appear  .
I confirmed the the hardware is okay , tested on marlin 2.0.6 work fine .
now 2.0.6 is crash,after platform IO update .
when 2.0.6 Bed Leveling all number under 0.1 but now 2.0.7 greater than 1 

### Hardware
Kossel delta printer with Hall probe sensor

### here is the probe result
M111 S32
G28
G29

Probe Z:-2.35 Discrepancy:-0.05
<<< Probe::run_z_probe  X-10.00 Y0.00 Z-2.35
>>> do_blocking_move_to  X-10.00 Y0.00 Z-2.35
>  X-10.00 Y0.00 Z12.65
  destination= X-10.00 Y0.00 Z-2.35 : destination = current_position
  destination= X-10.00 Y0.00 Z12.65 : prepare_fast_move_to_destination
  current_position= X-10.00 Y0.00 Z12.65 : z raise move
  current_position= X-10.00 Y0.00 Z12.65 : xy move
<<< do_blocking_move_to  X-10.00 Y0.00 Z12.65
<<< Probe::probe_at_point  X-10.00 Y0.00 Z12.65
>>> Probe::probe_at_point  X-10.00 Y0.00 Z12.65
...(50.00, 0.00, raise, 0, probe_relative)
  current_position= X-10.00 Y0.00 Z12.65 :
>>> do_blocking_move_to  X-10.00 Y0.00 Z12.65
>  X40.00 Y0.00 Z12.65
  destination= X-10.00 Y0.00 Z12.65 : destination = current_position
  current_position= X40.00 Y0.00 Z12.65 : xy move
<<< do_blocking_move_to  X40.00 Y0.00 Z12.65
  current_position= X40.00 Y0.00 Z12.65 : Probe::set_deployed
deploy: 1
>>> Probe::run_z_probe  X40.00 Y0.00 Z12.65
>>> Probe::probe_down_to_z  X40.00 Y0.00 Z12.65
>>> do_blocking_move_to  X40.00 Y0.00 Z12.65
>  X40.00 Y0.00 Z-3.00
  destination= X40.00 Y0.00 Z12.65 : destination = current_position
  current_position= X40.00 Y0.00 Z12.65 : xy move
  destination= X40.00 Y0.00 Z-3.00 : prepare_fast_move_to_destination
  current_position= X40.00 Y0.00 Z-3.00 : z lower move
<<< do_blocking_move_to  X40.00 Y0.00 Z-3.00
  current_position= X40.00 Y0.00 Z-0.70 : sync_plan_position
<<< Probe::probe_down_to_z  X40.00 Y0.00 Z-0.70
1st Probe Z:-0.70
>>> do_blocking_move_to  X40.00 Y0.00 Z-0.70
>  X40.00 Y0.00 Z14.30
  destination= X40.00 Y0.00 Z-0.70 : destination = current_position
  destination= X40.00 Y0.00 Z14.30 : prepare_fast_move_to_destination
  current_position= X40.00 Y0.00 Z14.30 : z raise move
  current_position= X40.00 Y0.00 Z14.30 : xy move
<<< do_blocking_move_to  X40.00 Y0.00 Z14.30
>>> Probe::probe_down_to_z  X40.00 Y0.00 Z14.30
>>> do_blocking_move_to  X40.00 Y0.00 Z14.30
>  X40.00 Y0.00 Z-3.00
  destination= X40.00 Y0.00 Z14.30 : destination = current_position
  current_position= X40.00 Y0.00 Z14.30 : xy move
  destination= X40.00 Y0.00 Z-3.00 : prepare_fast_move_to_destination
  current_position= X40.00 Y0.00 Z-3.00 : z lower move
echo:busy: processing
<<< do_blocking_move_to  X40.00 Y0.00 Z-3.00
  current_position= X40.00 Y0.00 Z-0.64 : sync_plan_position
<<< Probe::probe_down_to_z  X40.00 Y0.00 Z-0.64
2nd Probe Z:-0.64 Discrepancy:-0.06
[... 260 lines cut ...]
<<< do_blocking_move_to  X-10.00 Y100.00 Z19.48
<<< Probe::probe_at_point  X-10.00 Y100.00 Z19.48
  current_position= X-10.00 Y100.00 Z19.48 : Probe::set_deployed
deploy: 0
>>> do_blocking_move_to  X-10.00 Y100.00 Z19.48
>  X-10.00 Y100.00 Z19.48
  destination= X-10.00 Y100.00 Z19.48 : destination = current_position
  current_position= X-10.00 Y100.00 Z19.48 : xy move
<<< do_blocking_move_to  X-10.00 Y100.00 Z19.48
  current_position= X-10.00 Y100.00 Z19.48 : > probing complete
Bilinear Leveling Grid:
      0      1      2      3      4
 0 +5.765 +5.736 +6.080 +3.544 +3.503
 1 +5.565 +2.178 +0.008 +1.711 +4.321
 2 +5.988 -0.128 -2.372 -0.666 +5.004
 3 +3.214 +1.449 -0.878 +0.815 +2.269
 4 +2.806 +3.578 +4.460 +1.990 +1.032
G29 uncorrected Z:19.48
 corrected Z:15.20
  current_position= X-10.00 Y100.00 Z15.20 : sync_plan_position

### Configuration Files
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5456514/Configuration.zip)

### Did i miss something in configuration file ?

sjasonsmith, 2020-10-29:
Have you performed a G33 auto-calibration?

gellary, 2020-10-29:
yes i did, DELTA_ENDSTOP_ADJ and DELTA_TOWER_ANGLE_TRIM will affect the result 

> Have you performed a G33 auto-calibration?

gellary, 2020-11-02:
i find out the problem, every time finish G33 , i didn't use M500 
marlin 2.0.7 not problem 

marlin 2.0.7.2 "BABYSTEP_ALWAYS_AVAILABLE" will cause restart when adjust z probe offset .

boelle, 2020-11-22:
> marlin 2.0.7 not problem

So not a problem after all ?

gellary, 2020-11-24:
@boelle 
yes  2.0.7，not a problem

closed 2020-11-24 by gellary