MarlinFirmware/Marlin #19941: [BUG] Color UI - touching Ok button when in nozzle/bed/flow/speed adjustment causes to fire touch calibration

bkiziuk, 2020-10-29:
I have just moved to the bugfix of 2020-10-29.

### Bug Description

If the nozzle/bed temp or flow/speed settings are chosen by touching an icon on the status screen, the Ok icon on the setting page causes to execute touchscreen calibration.
It does not happen if the nozzle/bed temp adjustment is chosen from settings menu.

### Configuration Files

[bugfix-2-0-x_20201029.zip](https://github.com/MarlinFirmware/Marlin/files/5457241/bugfix-2-0-x_20201029.zip)

### Steps to Reproduce

1. touch nozzle icon on the status screen, the nozzle temp setting page should appear
2. touch Ok button, the calibration screen asking to touch left/right/top/bottom corners should appear

**Expected behavior:**

after touching Ok, the set temp/speed should be saved

**Actual behavior:**

touch calibration screen appears

#### Additional Information

Board: MKS Robin Nano 1.2 with stock TFT35 touch screen

rhapsodyv, 2020-10-29:
Sorry, I could not reproduce it. Are you sure that you are using the last bug fix? Can you try a clean build?

bkiziuk, 2020-10-29:
I cleaned up the install using Clean All in PlatformIO and rebuilded and reflashed the firmware. Then initiated the EEPROM to be sure that no old settings interfere but the issue seems to persist.

However, I played with it a bit and it seems to be somewhat random
[https://drive.google.com/file/d/1B0QAqR650A7vo3JeVVewIXo_egHNNmhH/view?usp=sharing](url)
I haven't noticed this kind of behaviour before updating to today's bugfix.

This is the current build that I'm using: [https://github.com/bkiziuk/Marlin/tree/124c5661e8d3ad604ecadd4ef83d6a4b97a469c3](url)

rhapsodyv, 2020-10-29:
Very weird. Can you try build the bugfix, not your tree? it could be a merge issue. 

I will do some tests.

bkiziuk, 2020-10-29:
I cloned a fresh copy of bugfix-2.0.x branch, loaded my config files (Configuration.h, Configuration_adv.h, platformio.ini, pins_MKS_ROBIN_NANO.h) and built it. The issue persisted.

Then, I came back to my own branch and reverted the last merge [https://github.com/bkiziuk/Marlin/pull/7](url)
```git revert -m 1 124c5661e8d3ad604ecadd4ef83d6a4b97a469c3```

The issue seems to be gone.
So, it looks like there are some changes involved that were made within last 14 days after [https://github.com/bkiziuk/Marlin/pull/6](url) was created.

rhapsodyv, 2020-10-29:
I could not reproduce it yet. But, I think I found the possibly cause.

rhapsodyv, 2020-10-29:
Can you test if this fixes the issue for you: #19951 ?
Thanks

bkiziuk, 2020-10-29:
I made manual modification to my branch and tested it.
Unfortunately it does not solve an original issue.
There is also an additional side effect of unnatural (double or tripple) short "chirp" when touching a button when the correct action is performed and a "normal" one otherwise. Fans are bit noisy but hopefully it can be heard.
[https://drive.google.com/file/d/1B2o5MAxwTrrtGCAfxmtRePsrwl4fqBNO/view?usp=sharing](url)

rhapsodyv, 2020-10-29:
Hum... I will test in in a nano v1.2.

rhapsodyv, 2020-10-31:
Can you test it again?

bkiziuk, 2020-10-31:
Anything in particular?
I have just cloned the latest bugfix-2.0.x branch and I can still see the issue - the same as it was originally reported.

rhapsodyv, 2020-10-31:
Test in my PR that fix it: https://github.com/MarlinFirmware/Marlin/pull/19951

bkiziuk, 2020-10-31:
Tested with the bugfix-2.0.x branch I cloned above. 
Looks good. I can not observe the issue any more.

rhapsodyv, 2020-11-01:
The fix for this was merged. It can be closed.

bkiziuk, 2020-11-01:
Confirmed. Thank you.

PR #19951 (2020-11-01): Reset calibration screen touch timer when click is consumed 
### Description

With recent changes to adapt color ui for `wait_for_release`, we may break a timer to start touch calibration screen if the touch is held for more than 2.5s. This PR fix that timer.

### Benefits

Fix: #19941 

### Related Issues

#19771
closed 2020-11-01 by bkiziuk