MarlinFirmware/Marlin #19943: [BUG] Rough curved surfaces

Testmangh, 2020-10-29:
### Bug Description

My prints were perfect but 17~18 days ago, after a update (bugfix), all curved surfaces are rough.
I tried reinstalling marlin (bugfix), installing the stable 2.0 version, disabling features that consumes processor cycles, tried to run Octoprint on safe mode and printing from SD card, tried enabling S Curve and classic jerk, tried printing slower and some other things...  But nothing fixed the problem...

My next try would be to change the motors stepper drivers and motherboard, but maybe u guys could help me.

All the tries above were done yesterday because 17~16 days ago I had to travel.

### Configuration Files

[https://drive.google.com/drive/folders/1OL8hEnxIRKYpsGyHJeYMojSav0ic4_Zp?usp=sharing](url)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

I don't know what I exactly did, but I just updated the firmware about 18 days ago.

**Expected behavior:**

Smooth curved surfaces.

**Actual behavior:**

Rough curved surfaces.

#### Additional Information

My prints 18 days ago
[https://i.imgur.com/m6msyms.jpg](url)
[https://i.imgur.com/EHi58cB.jpg](url)

My prints now
[https://i.imgur.com/CvRX2oy.jpg](url)
[https://i.imgur.com/SaU1Z8w.jpg](url)

I also can see the printer making some small pauses while printing curves.

thordarsen, 2020-10-29:
What slicer are you using? This sounds like the issue many people are having with Cura 4.7 ( https://github.com/Ultimaker/Cura/issues/8321 ) 
I had the same issue which was resolved by downgrading back to 4.6

Testmangh, 2020-10-29:
Yep, cura 4.7 I'm gonna downgrade and test to see if it fixes it... Thanks  a lot man!

Testmangh, 2020-10-29:
It got a little better but not as good as before, im gonna try cura 4.5

Testmangh, 2020-10-30:
Ok, thanks @thordarsen Cura 4.5 did it... Now I have perfect prints again.

closed 2020-10-30 by Testmangh