MarlinFirmware/Marlin #19960: [FR] [W.I.P.]  Expanded Long_File_Name Support - AKA Allow M20 to report long file-names as default

RFBomb, 2020-10-30:
### Description

I opened up a reddit discussion here as well:
https://www.reddit.com/r/MarlinFirmware/comments/jkurc8/updating_firmware_to_use_longfilenames_can/

I got tired of Octoprint displaying the DOS8,3 filename gibberish when reading and writing to the SD card. This occurs even with The ‘host uses long file names’ define turned on. So I decided to modify the code accordingly. I'm coming here to discuss it before implementation via pull request.

### Feature Workflow

Octoprint reads the files on the SD card using M20. Currently, M20 reports all filenames as the gibberish that is Dos 8.3 filenames. I understand why its used (smaller memory footprint mostly), but it would be nice to have it display long file names / paths by default. So i set out to introduce that feature. 

1. I have modified CardReader.cpp and CardReader.h in order to have some boolean parameters passed into the following routines:
  - printListing
  - ls
  - printLongPath

2. I have modified the M20 GCode command to call out 'Card.ls(true)' or 'Card.ls(false)' depending if the longfilename support define is turned on or not. Currently, I have a secondary define created to turn the feature to default. I would like to have it be able to accept 'M20 L' and 'M20 S' to request short/long names on demand (with M20 L only working if LongFileName is defined) I currently don't know how to code for that, so its only mocked in and commented out. 

3. Other GCode commands that I know interact with the SD Card that I have not modified yet, but may need to be looked at in order to work with this functionality. 

- Remove File (M30)
> -     Will check functionality by using Octoprint's "FileManager" plugin to see how it sees/attempts to delete files on SD card.
> -     May have to check for long_file_name in M30 command and convert back to Dos filename to work properly.
> -     Must look at octoprint terminal to determine how it sends the request

- Write File (M28)
> -     Will check by enabling the new functionality and uploading a print file to the SD card via Octoprint.
> -     See notes 2+3 for m30 command

- Select File (M23)
> -     See notes 2+3 for m30 command

- Select and Start (M32)
> -     Do people even use this feature?!
> -     Even so, this would be highly customized for each printer's SD card. Probably not worth looking at. But if needed, could use likely use similar results as the other functions noted above for LongFileName functionality.

#### Additional Information
I have uploaded several files I've already modified. The M20 command works great to retrieve the long filenames. I haven't tested any other functions yet though. I have listed all the functions I believe need to be modified / examined for this functionality. If there are others, I'm unaware of them currently. 
[Proposed Changes So Far.zip](https://github.com/MarlinFirmware/Marlin/files/5465333/Proposed.Changes.So.Far.zip)

After some more testing, I plan to create a pull request with my proposed changes. I just wanted to open up a discussion here on the idea before I do that.

Bringing this functionality to things like reading off of the USB / SD_Card2 is something that may have to be looked into. I am running an SKR 1.3 board, so I will be unable to test any of that functionality.

ellensp, 2020-10-30:
you might want to have a read of https://github.com/MarlinFirmware/Marlin/pull/19111

RFBomb, 2020-10-30:
Ok, so what I gather from that is:

- Octoprint will never support Marlin's recomendation of using M33 to parse the filenames due to possible (likely) lockup scenario. 
- Marlin doesn't want to implement a feature like this because too many language / special character variations in possible 'longnames'. It would be a move from 'reliable yet cumbersome (for humans) file navigation' to support pretty filenames for neglible benefit. 

Also, Marlin doesn't want to implement an optional feature like this with my proposed M20 L / S (which is exactly what the other post suggested) since they would rather do an overhaul than a one-off. And due to the above note, the overhaul will never happen. 

Looks like my code will be in my personal build then.

ellensp, 2020-10-30:
I was toying with simply encoding long names as as a hex string. Then client can decode and handle language issues how they want...

RFBomb, 2020-10-30:
That should work. It would be up to the clients to then handle the decoding.
I think all the clients would have a hard time arguing about which way to use (request long file name natively, or decode the hex on client-side). That would make all clients more complicated to add that in I think (though I'll admit I don't know anything about decoding hex)

My solution is doing exactly what the marlin devs recomend - It calls the M33 function. The only difference here is instead of returning the DOS command, then having the client call M33 for each line (which would require the client having a database of all the dos names correlated to the long names); I am doing it natively. As in I retrieve the Dos name, then just run the M33 conversion prior to sending it back over serial. It knocks the client reaction out of the equation.

RFBomb, 2020-11-01:
Closing issue since I've opened up a pull request. Disussion ought to continue on that pull request, since code is already written and tested.

closed 2020-11-01 by RFBomb