MarlinFirmware/Marlin #19964: [BUG] Multiple Probing and Probe Extra doesn't work for homing with z probe

yysh12, 2020-10-31:
### Bug Description
Not sure if bug or intentional by design, but:

When:
`#define USE_PROBE_FOR_Z_HOMING` //(with z safe homing)

And:
`#define MULTIPLE_PROBING 2`
`#define EXTRA_PROBING    1 `

G29 will probe each point thrice, 
but G28 will only probe twice, once fast and once slow. 

I have inaccurate probe and hence need for extra probing. This occasionally leads to wrong homing z value without the extra probing. Is there any reason G28 doesn't utilize extra probing parameters?

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5468467/Configuration.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1.) G28

**Expected behavior:**

1) G28 probes thrice, averaging two points and discarding one

**Actual behavior:**

G28 only probes twice, occasionally leading to erroneous value. 

#### Additional Information

I could reduce speed of the probe, but that just lengthens the entire leveling sequence beyond 10 minutes. Ideally just an extra homing probe would do the trick.

thisiskeithb, 2020-10-31:
This isn't a Marlin bug.

As their name implies `MULTIPLE_PROBING` and `EXTRA_PROBING` is for probing, not homing. Homing Z will home, backoff, then home again.

> I could reduce speed of the probe, but that just lengthens the entire leveling sequence beyond 10 minutes. Ideally just an extra homing probe would do the trick.

`Z_PROBE_SPEED_FAST` and `Z_PROBE_SPEED_SLOW` are configurable, so you can increase their values. But if your probe is inaccurate, it'd be better to replace it/track down why.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-10-31 by thisiskeithb

yysh12, 2020-10-31:
Ah gotcha. Thanks. That makes sense.

Closed #19964<https://github.com/MarlinFirmware/Marlin/issues/19964>.