MarlinFirmware/Marlin #19966: [BUG] Z-Axis quad motor, 4th motor does not stop when hits endstop

PKav89, 2020-10-31:
### Bug Description

I am building large (about 2.5 meters tall) cartesian machine with 4 Z axis steppers at the corners. I am using FYSETC F6 v1.3 board (ATMega2560) to control it. I have configured Marlin with NUM_Z_STEPPER_DRIVERS 4 to use Z, E0, E1 and E2 drivers as Z1, Z2, Z3, and Z4 motors, and Z_MULTI_ENDSTOPS to use sepparate endstop for each corner of the machine. When I home it with G28 Z gantry goes up, Z1, Z2 and Z3 hits endstop and stops, but Z4 motor hits endstop and keeps rotating. All endstops are set correctly, M119 command says, that all endstops reacts correctly when hit. Tested with 2.0.7.2 and bugfix, issue still exists.

### Steps to Reproduce

Configure , build and upload Marlin to the board
Home Z axis with G28 Z

**Expected behavior:**

All four motors hit their respective endstops and stops

**Actual behavior:**

Z4 motor keeps rotating even after hits endstop.

#### Additional Information

I also modified Marlin\src\pins\ramps\pins_FYSETC_F6_13.h file. Z_MAX_PIN was set to 9, but should be 2. Tested with real board, all endstop pins are now correct.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5469125/Configuration.zip)

sjasonsmith, 2020-10-31:
I looked through the code and didn't see anything obviously wrong with Z4. @InsanityAutomation, had you tested with 4 endstops when you added Quad-Z?

PKav89, 2020-11-01:
I can program MCUs in C and I have AVR Dragon debugger, so I can try to debug if you show me where, because Marlin source looks too complicated for me to understand at once. :)

boelle, 2020-11-01:
Z it set to home toward max

// Direction of endstops when homing; 1=MAX, -1=MIN
#define Z_HOME_DIR +1 (it should be 1 and not +1)

so endstop for Z1 motor will be Z
endstop for Z2 will be Z2
....
endstop for Z4 should be Z4

what does M119 say when each endstop is triggered and not triggered? does it say anything for Z4 ?

boelle, 2020-11-01:
and is the endstops connected like this?

Z1 ZMAX
Z2 XMAX
Z3 YMAX
Z4 ZMIN

PKav89, 2020-11-02:
@boelle Yes, M119 shows all four Z endstops and shows their open/TRIGGERED condition correctly.
And yes, endstops wired as in Configuration_adv.h.

sjasonsmith, 2020-11-07:
@PKav89 does the 4th motor eventually stop, or does it grind until you turn off power?

PKav89, 2020-11-07:
@sjasonsmith I left it running for about 20 minutes and after that it kept rotating.

PKav89, 2020-11-13:
So I found the main reason of the issue. It is hardcoded somewhere to work with Z homing to MIN.
If I set Z_HOME_DIR to -1 and invert Z direction, everything works as expected and machine homes correctly, but if I set Z_HOME_DIR to 1 endstops does not work.

Also, if Z_HOME_DIR set to -1, M119 returns:

>  M119
> SENDING:M119
> Reporting endstop status
> x_min: open
> y_min: open
> z_min: open
> z2_max: open
> z3_max: open
> z4_max: open
> ok P15 B3

M119 says, that Z2, Z3 and Z4 are expected to trigger at MAX, which is correct, but Z1 endstop is expected to trigger at MIN, which is wrong. And when Z_HOME_DIR set to -1, M119 returns correct information:

> > M119
> SENDING:M119
> Reporting endstop status
> x_min: open
> y_min: open
> z_min: open
> z2_min: open
> z3_min: open
> z4_min: open
> ok P15 B3

Hope it helps to fix this issue.

sjasonsmith, 2020-11-15:
>z_min: open
>z2_max: open
>z3_max: open
>z4_max: open

I was not able to see in the code how that would happen with your configuration files, and couldn't reproduce it myself. I couldn't use your config exactly as-is, but was able to load something extremely similar in a simulated environment. Everything shows `max` as you would expect.

The code I was running was on top of commit 0465e0ae3aaec257618d095f7e2b28ef677dbe4a, which is 5 days old.

Are you able to confirm that M119 behavior with the latest bugfix commits?

closed 2020-12-23 by github-actions[bot]