MarlinFirmware/Marlin #19967: [FR] Servo control by M3-M5 commands

spin7ion, 2020-10-31:
### Description

There is class of pcb cnc mills that use RC motor and RC ESC as spindle. It's good to have servo controlled by commands M3-M5

### Feature Workflow

M3 S*** controls pwm of one of servo
M5 sets servo to minimal angle

spin7ion, 2020-10-31:
To do so make such changes:
M3-M5.cpp
```
void GcodeSuite::M3_M4(const bool is_M4) {
  #if EITHER(SPINDLE_FEATURE, SPINDLE_SERVO)
    planner.synchronize();   // Wait for movement to complete before changing power
  #endif

  #if ENABLED(SPINDLE_SERVO)
    servo[SPINDLE_SERVO_ID].move(parser.intval('S', SPINDLE_SERVO_MIN));
  #endif

  cutter.set_direction(is_M4);

  #if ENABLED(SPINDLE_LASER_PWM)
    if (parser.seenval('O'))
      cutter.set_ocr_power(parser.value_byte()); // The OCR is a value from 0 to 255 (uint8_t)
    else
      cutter.set_power(parser.intval('S', 255));
  #else
    cutter.set_enabled(true);
  #endif
}

void GcodeSuite::M5() {
  #if EITHER(SPINDLE_FEATURE, SPINDLE_SERVO)
    planner.synchronize();
  #endif

  #if ENABLED(SPINDLE_SERVO)
    servo[SPINDLE_SERVO_ID].move(parser.intval('S', SPINDLE_SERVO_MIN));
  #endif
  
  cutter.set_enabled(false);
}

#endif
```
Configuration_adv.h
```
#ifdef SPINDLE_SERVO
  #define SPINDLE_SERVO_ID 0
  #define SPINDLE_SERVO_MIN 10
#endif
```

sjasonsmith, 2020-10-31:
Do you already have this working?

Can you provide a link to a machine that uses this?

spin7ion, 2020-10-31:
Yes I do, my CNC uses this: [Link to build](https://pikabu.ru/story/frezernyiy_stanok_dlya_pechatnyikh_plat_7033156) 
Moreover ANT project utilizes same idea: [Ant](https://www.youtube.com/channel/UCX44z-SSL7LzcB4xxgUdHHA)

sjasonsmith, 2020-10-31:
Can you explain why you need to control this through the servo output, rather than through the PWM spindle support which already exists? Given that Servo output is really just PWM, it seems like it would already be possible, although support and configuration may be very hardware-dependent.

I am not saying your change is unnecessary, I am simply trying to better understand the change.

sjasonsmith, 2020-10-31:
I do think it would be useful for you to post a pull request with the suggestion, assuming you have good reasons the existing PWM Spindle support is insufficient.

I think I am missing some things when I compare your code snippets to the current M3_M4 function. Seeing your changes within the current code (in a PR) would be very helpful.

The largest question I have right now is whether this works _with_ the current `SpindleLaser` class, or if you are using the servo control to _replace_ the SpindleLaser class.

spin7ion, 2020-10-31:
Servo PWM is kind of special, please check the provided picture:
[image]
I can't find any prove that PWM spindle support can meet the requirements

spin7ion, 2020-10-31:
I just made some additions to M3-M5 command algorithm. You can check the code in my initial code, but it is for older Marlin
[This is link to the video of working servo spindle](https://youtu.be/ZwUy9kIP4Gw)

sjasonsmith, 2020-10-31:
I see, so in the "hobby servo" world, the actual pulse width is more important than the duty cycle. Half the pulse width at twice the rate (same duty cycle) would not work?

It seems like a valid feature to me, I'd like to see the pull request, to see how it integrates with the other Spindle/Laser features.

spin7ion, 2020-10-31:
Please, check out my [pull request](https://github.com/MarlinFirmware/Marlin/pull/19968). Haven't done any yet, so I will be happy to any advice.

spin7ion, 2020-10-31:
Ok, now I hopefully made some good changes. Now feature utilizes spindle class

ManuelMcLure, 2020-10-31:
How does this differ from the signal generated by `M280`?

sjasonsmith, 2020-10-31:
I'm just adding a link to the closed request which was incorrectly targeting the 2.0.x branch. I have not looked at the new changes yet to determine whether my concerns were all resolved.
#19968

spin7ion, 2020-10-31:
> How does this differ from the signal generated by `M280`?

It is not differ. Actually it use same servo class as M280. The main idea is to allow M3-M5 commands to drive the spindle made of RC hobby brushless motor and ESC. So g-code will be compatible with different CNCs.

closed 2020-11-01 by thisiskeithb