MarlinFirmware/Marlin #19969: [BUG] Filament runout does not trigger M600

JanKolibri, 2020-10-31:
### Bug Description

Filament sensor does not respond.
State of the sensor was registered (M119). However M600 was not triggered.
Galvanic connection was checked.

Feedback 'M119':
No Filament --> 'filament: open'
Filament present --> 'filament: triggered'

### Configuration Files

Attachement:
Configuration.h
Configuration_adv.h
platformio.ini
Layout_BTT_SKR_V1_3.png

Modifications made `(unrelated):`
Nozzlecleaner added at 'Y+' (5V, Pin1_26, GND).

Activated direct PIN_CONTROL (Marlin\Configuration_adv.h\#define DIRECT_PIN_CONTROL).

Movement of Nozzle cleaner declared.
#define NOZZLE_CLEAN_START_POINT { { 225, 0, current_position.z + NOZZLE_PARK_Z_RAISE_MIN } }
#define NOZZLE_CLEAN_END_POINT   { { 245, 0, current_position.z + NOZZLE_PARK_Z_RAISE_MIN } }

### Steps to Reproduce

Used Marlin: 2.0.7.2 (2020-10-31)
Mainboard: BTT SKR v1.3 (see annotation Layout)
Runout Sensor: P1_28 + GND
Type: DIY
Galvanic connection: √

**Step 1:** Configuration of sensor (unsuccessful)

_____Configuration_adv.h____
#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_ENABLED_DEFAULT true 
  #define NUM_RUNOUT_SENSORS   1          
  #define FIL_RUNOUT_STATE     LOW        
  #define FIL_RUNOUT_PULLUP                
 //#define FIL_RUNOUT_PULLDOWN           

 #define FILAMENT_RUNOUT_SCRIPT "M600"
 #define FILAMENT_RUNOUT_DISTANCE_MM 5

  #ifdef FILAMENT_RUNOUT_DISTANCE_MM
       //#define FILAMENT_MOTION_SENSOR
  #endif

___pins_BTT_SKR_V1_3.h____
 #define FIL_RUNOUT_PIN                   P1_28

**Step 2:** Change of State (unsuccessful)
#define FIL_RUNOUT_STATE     LOW --> HIGH 

**Step 3:** Change of State (unsuccessful)
//#define FIL_RUNOUT_PULLUP                
 #define FIL_RUNOUT_PULLDOWN

**Expected behavior:**

Triggering 'M600' when state of runout sensor is being changed

**Actual behavior:**

absent 'M600' when runout is being triggered

#### Additional Information

Thank you for your help in advance (ツ)╯

rhapsodyv, 2020-10-31:
How are you printing?

Runout will only trigger if and only if one of this 3 criteria are met:

1 - You are printing using SD (sd for marlin, not SD from a serial TFT)
2 - You send M75 to start the printer job timer
3 - Your gcode have M109/M119 (wait target temperature)

Do you met any of these criteria?

JanKolibri, 2020-10-31:
That fixed it. ᕙ(^‿^)ᕗ
Thank you for the quick reply

For 1: I am printing with Octoprint. No SD.
For 2: I did not include 'M75' in my starting script. This though fixed it.
For 3: No - I didn't include it, because the nozzle was being heated during autobed leveling.

Could please explain the 'M75' command?

qwewer0, 2020-10-31:
[M75 - Start Print Job Timer](https://marlinfw.org/docs/gcode/M075.html)

JanKolibri, 2020-10-31:
Thank you for the link. 
I checked this one already, but it doesn't give much information, which is why I am asking.

closed 2020-10-31 by thisiskeithb