MarlinFirmware/Marlin #19974: [BUG] Commit TMC Interpolation per driver breaks HYBRID_THRESHOLD

BryanSmithDev, 2020-10-31:
### Bug Description

Ever since commit 4fccb92e072463bd6147b2ce22352b79337cb53c having HYBRID_THRESHOLD enabled in configuration_adv has caused my BTT SKR1.3 to bootloop at the marlin boot screen. 

### Configuration Files

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5469846/Config.zip)

### Steps to Reproduce

1. Start with current bugfix-2.0.x branch that contains 4fccb92e072463bd6147b2ce22352b79337cb53c
2. Uncomment #define HYBRID_THRESHOLD
3. Compile, it should succeed without issue.
4. Attempt to boot.
5. Observe bootloop

On my BTT SKR1.3 with TMC2130s it will bootloop unless commit 4fccb92e072463bd6147b2ce22352b79337cb53c is reverted, OR HYBRID_THRESHOLD is not enabled.

**Expected behavior:**

Marlin boots as intended.

**Actual behavior:**

Marlin boot loops, showing the Marlin splash screen over and over infinitely.

#### Additional Information

Apparently, I had Hybrid Threshold enabled from a while back and spent quite a while tracking down which commit caused the boot looping until I determined the commit and then narrowed it down from there. I am not sure what part of that commit would have affected it (maybe changes or missed changes to trinamic.cpp?). I do not need hybrid threshold currently so disabling it works for me. But I figured if there is something amiss I would report it.

sjasonsmith, 2020-11-07:
When I look at the code I don't see any obvious issues. Tagging @ManuelMcLure since they added the change mentioned above.

ManuelMcLure, 2020-11-07:
That’s odd. The behavior if you don’t uncomment any of the specific driver interpolation settings should be identical to what it was before except for the output of `M122` - the new code doesn’t even store it in EEPROM since it can’t be changed at runtime, only at compile time. I don’t have an SKR 1.3 but I do have a Re-ARM so I’ll test it on that in a day or so.

BryanSmithDev, 2020-11-07:
@sjasonsmith @ManuelMcLure I will close this issue. I wanted to triple check myself on a fresh day, and in order to help eliminate any potential issues related to other configurations I had, I started with a clean checkout at the above specified commit (which I should have done prior to making the issue). I then only changed minimally the configuration.h and configuration_adv.h to specify my board, drivers, and setup serial ports. I then compiled and flashed and it booted with no problems. I then uncommented hybrid threshold and compiled and flashed and it also booted just fine. I am not sure what else must have been at play with my full configuration to have caused the boot loops but that is an issue likely of my own making or there was some interplay with other features I had enabled initially. Either way, I do not think this issue is needed anymore. My apologies for taking your time.

closed 2020-11-07 by BryanSmithDev

sjasonsmith, 2020-11-07:
Thanks for the follow-up @BryanSmithDev. If you figure out what caused your issue feel free to ping us on Discord, or post a new bug report if you are certain something is misbehaving.