MarlinFirmware/Marlin #19980: [BUG] SWITCHING_EXTRUDER_E23_SERVO_NR 2

DrumClock, 2020-11-01:
**Printer configuration:**

- stepper motor E0 for the extruder E0, E1
- stepper motor E1 for the extruder E2, E3
- 4 hotends (independent) on one carriage
- Servo0 is reserved for the BLTouch Z-probe

```
Board: 		BTT GTR V1.0 + M5 V1.0
Controller:	LCD Full graphic 128x64 pix.
Marlin:         bugfix-2.0.x  distribution date 2020-10-30
Compiled:	Visual Studio Code 
```

**Actual behavior:**

If I use G-code T0, T1, T2, T3:
- T0 and T1 = servo 1 sets the angles for E0 / E1.
- T2 and T3 = servo 2 sets angles of 0/180 degrees.
  It basically doesn't matter what is defined for E2 / E3.

If I use G-code M280 P1 S45 or M280 P2 S45,
Sets both servos to the specified angle.

**Expected behavior:**

When using G-code T2, T3 sets servo 2 to the correct hives
as defined for E2 / E3.

**Major changes:**

configuration.h
**********************************
```
#define EXTRUDERS 4
#define HOTENDS 4
.
.
#define SWITCHING_EXTRUDER
#if ENABLED (SWITCHING_EXTRUDER)
  #define SWITCHING_EXTRUDER_SERVO_NR 1
  #define SWITCHING_EXTRUDER_SERVO_ANGLES {45, 135, 45, 135}
  #if EXTRUDERS> 3
    #define SWITCHING_EXTRUDER_E23_SERVO_NR 2
  #endif
#endif
.
.
.
#define NUM_SERVOS 3
#define SERVO_DELAY {300, 500, 1000}

```
pins_BTT_GTR_V1_0.h
**********************************

```
#define SERVO1_PIN PI8
#define SERVO2_PIN PI10

```

Conditionals_LCD.h
***********************************
```
#if DISABLED(SWITCHING_NOZZLE)
    //#define HOTENDS      E_STEPPERS
 #endif
```

Here is the whole FW of this modification, including the configuration.
https://github.com/DrumClock/Marlin/tree/GTR-2xMFG-V7

https://github.com/DrumClock/Marlin/archive/GTR-2xMFG-V7.zip

Can you help me with that?

boelle, 2020-11-02:
Sticking my neck out here, is this a bug ?

DrumClock, 2020-11-02:
Hi @boelle 
Yes, is it a mistake?
Servo 2 for SWITCHING_EXTRUDER_E23_SERVO_NR does not work properly.

boelle, 2020-11-02:
no i was just confused if it was a bug or a question

but a carefull guess or something you can test is #define NUM_SERVOS 3, should that not be 2?

i only have the real bl touch and i dont have servo's defined and my thought is that the bl touch goes on the first servo defined in the pins file if no servo is enabled

DrumClock, 2020-11-02:
mom set #define NUM_SERVOS 3 because:

Servo0 is reserved for the BLTouch Z-probe
Servo1 switches extruders E0, E1
Servo2 switches extruders E2, E3

It is currently servo2 that does not switch according to the settings:
#define SWITCHING_EXTRUDER_SERVO_ANGLES {45, 135, 45, 135}

boelle, 2020-11-02:
and i have //#define NUM_SERVOS 3 

bl touch connected to servo0

so just as an experiment set it to #define NUM_SERVOS 2

bltouch will be either on servo0 or servo2

DrumClock, 2020-11-02:
I'm not dealing with a servo for BL Touch, that's fine on the first servo (servo0).

SWITCHING_EXTRUDER does not work for the third servo.
The second servo (servo1) switches the extruder E0, E1 in order.
The third servo (Servo2) switches the extruder E2, E3 at the wrong angles.

See description above.
Should I make a video?

boelle, 2020-11-02:
> Should I make a video?

yes and try #define NUM_SERVOS 2

DrumClock, 2020-11-02:
yes i will try and send the video but I probably won't make it until tomorrow today.

DrumClock, 2020-11-02:
#define NUM_SERVOS 2 doesn't solve anything.
Because servo0 is for BLTouch and servo1 is for extruder E0, E1 and servo2 is not active at this time.

Tomorrow I will send a video with the original setting #define NUM_SERVOS 3
I physically need 3 servos

boelle, 2020-11-02:
oki it was just a thought since bltouch works without #define NUM_SERVOS enabled

DrumClock, 2020-11-02:
Servo2 which switches extruder E2, E3 is active. 
It just doesn't set to the specified 45/135 degree angles but at 0 and 180. 

I believe this is a bug in Marlin because the M280 P2 command is working properly.

DrumClock, 2020-11-02:
HI @boelle 
so i filmed it  https://youtu.be/adPkDCUyv7E
I sent G-codes via pronterface, see. captions in the video.

sjasonsmith, 2020-11-07:
@DrumClock I might have found an issue in the code. Can you try replacing the following code in tool_change.cpp? I just changed the single commented out line. I looks like it was incorrectly indexing into the 2-D array when retrieving the angles.

```
  void move_extruder_servo(const uint8_t e) {
    planner.synchronize();
    #if EXTRUDERS & 1
      if (e < EXTRUDERS - 1)
    #endif
    {
      // sjasonsmith modified the index at the end of the line
      //MOVE_SERVO(_SERVO_NR(e), servo_angles[_SERVO_NR(e)][e]);
      MOVE_SERVO(_SERVO_NR(e), servo_angles[_SERVO_NR(e)][e&1]);
      safe_delay(500);
    }
  }
```

DrumClock, 2020-11-07:
**The number of servos does not matter!**
I defined `NUM_SERVOS 2` and found this behavior:

- the **first servo** (SERVO0) works correctly according to the angle setting 35/145
 
- the **second servo** (SERVO1) always switches to end angles 0/180.
   the setting angles of the second servo are affected by the `SERVO_DELAY `value.
   the higher it is, the closer the servo approaches 0/180 end positions

sjasonsmith, 2020-11-07:
@DrumClock I don't see how that relates to the change I posted at all.

The code I replaced was using the tool number in a 2-element array. That is fine for tools 0/1, but for tools 2/3 it will either read a value from the next servo (if there is one), or it will read whatever the compiler has placed in the adjacent memory.

sjasonsmith, 2020-11-07:
It is possible that SERVO_DELAY is stored in the memory it was incorrectly accessing.

DrumClock, 2020-11-07:
Hi @sjasonsmith 
I will try to replace the code and write what has changed

DrumClock, 2020-11-07:
Hi @sjasonsmith 
Yes, the servo for E2, E3 is already in the correct end positions.
Only the values from` SWITCHING_EXTRUDER_SERVO_ANGLES {45, 135, 45, 135}` for E2, E3 are read in reverse.
 I.e :
 45 degrees is loaded for E3 instead of  E2
135 degrees is loaded for E2 instead of E3

sjasonsmith, 2020-11-07:
I don't see in the code how that could be possible. Perhaps they are mechanically reversed? Maybe you are looking at your servo from the wrong direction?

DrumClock, 2020-11-07:
I have the servos and motors connected as well.
switching E0, E1 and the direction of motor rotation is according to the settings for servo1.
switching E2, E2 and the direction of motor rotation are exactly opposite for servo2.

If I swap servos between each other, it behaves the same way.

DrumClock, 2020-11-07:
When I change `#define SWITCHING_EXTRUDER_SERVO_ANGLES {45, 135, 135, 45} // Angles for E0, E1 [, E2, E3]`

servo2 switches in the same direction as servo1 but the motor for E2, E3 rotates in reverse

sjasonsmith, 2020-11-07:
You could add something like this next to the line I modified, so that it will print the tool, servo, and angle every time you change tools. That might help identify what is wrong. (I didn't compile this, but I think it is correct)

`SERIAL_ECHOLNPAIR("Tool:", e, " Servo:", _SERVO_NR(e), " Angle:", servo_angles[_SERVO_NR(e)][e&1]);`

DrumClock, 2020-11-07:
Here is the listing:
```
>>> t0
SENDING: T0
Tool: 0 Servo: 1 Angle: 45
echo: Active Extruder: 0
>>> t1
SENDING: T1
Tool: 1 Servo: 1 Angle: 135
echo: Active Extruder: 1
>>> t2
SENDING: T2
Tool: 2 Servo: 2 Angle: 45
echo: Active Extruder: 2
>>> t3
SENDING: T3
Tool: 3 Servo: 2 Angle: 135
echo: Active Extruder: 3

```
the motor for E2, E3 rotates in the opposite direction to peo E0, E1

DrumClock, 2020-11-07:
sorry @sjasonsmith  i forgot this:

```
#define INVERT_E0_DIR false
#define INVERT_E1_DIR false
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false

```
**it's okay now**

sjasonsmith, 2020-11-07:
OK, so with just that one tiny code fix it is good?

DrumClock, 2020-11-07:
Switching and direction is fine when I test it via the terminal using G-codecs.
I'll try the real print.

sjasonsmith, 2020-11-07:
I had gotten a little confused. I didn't realized you switched from servos to extruder motor a few comments back. I'll post a pull request for the servo fix.

DrumClock, 2020-11-07:
OK thanks.

DrumClock, 2020-11-07:
The extruder motors E1,E2,E3 do not work properly.
After a few seconds, E1, E2 turn off and E3 does not rotate at all.

I don't know if it's a given configuration
`#define EXTRUDERS 4` and `#define HOTENDS 4`

sjasonsmith, 2020-11-07:
I don't know what might cause that. I have zero experience working with multiple extruders. Perhaps there are other configuration options which should impact that.

I would like to keep this issue focused on the servo angle issue which we fixed. I consider motor movement issues separate from this.

DrumClock, 2020-11-07:
OK, we have solved the servo angles, do I have to close the request?

sjasonsmith, 2020-11-07:
Wait for the fix to be merged.

DrumClock, 2020-11-07:
How do I know if the patch has been merged?

ElectroFlux-SS, 2020-11-07:
Disable the below line which is why E2 and E3 stop working shortly after printing (known or previously reported issue).
#define DISABLE_INACTIVE_EXTRUDER

DrumClock, 2020-11-07:
Hi @ElectroFlux-SS 
Yes this setting works.
`// #define DISABLE_INACTIVE_EXTRUDER`

boelle, 2020-11-07:
Problem soved then?

sjasonsmith, 2020-11-07:
@boelle thatbwas a distraction from the real issue reported here, which is waiting for a PR to merge.

DrumClock, 2020-11-08:
I confirm the functionality of Merged # 20060 into bugfix-2.0.x.

closed 2020-11-08 by DrumClock

ElectroFlux-SS, 2020-11-08:
That's all good new Petr.

I'm just a little confused about your extruder directions and did not want
to post on Marlin and confuse anyone.

So you have all your invert E directions set to false (INVERT_E0_DIR,
INVERT_E1_DIR, etc), however shouldn't these alternate as false, true,
false, true as every other extruder is needed to turn in the
opposite direction (T0=CCW, T1=CW, T2=CCW, T3=CW)?

DrumClock, 2020-11-08:
Yes, that's how I tried it now.
Just set poure true = CCW / false = CW for
#define INVERT_E0_DIR
#define INVERT_E1_DIR
Because the motors are connected to connectors E0 and E1

Marlin negates the speed for the required extruder motor.
for motor E0 there are extruders E0 and reverse E1
for motor E1 there are extruders E2 and reverse E3