MarlinFirmware/Marlin #19982: some problem on 2 times retract

aghfars, 2020-11-01:
i have problem on 0.2 mm nozzle and 0.1 mm layer height.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5470852/Marlin.zip)
Marlin 2.0.7.1
[image]
And I tried to fix this problem, which worked properly by deleting the original retract, but then it crashed due to other retractions.
the slicer is cura 4.7.1

boelle, 2020-11-01:
is this not a problem with cura that makes the Gcode?

aghfars, 2020-11-02:
> is this not a problem with cura that makes the Gcode?
cura make the Gcode and the Gcode seems to be correct
so why when i remove the first retract it works fine?
and i test the relative and absolute extruder and nothing change.

boelle, 2020-11-02:
i use relative extrusion.... 

so you remove an retract after G1 F1500 E0 ?? or are you removing G1 F1500 E0?

please attach not working code and the working one

but if all you change is the gcode then its cura that outputs wrong gcode and the problem is not related to marlin

aghfars, 2020-11-02:
[Working code.zip](https://github.com/MarlinFirmware/Marlin/files/5476295/Working.code.zip)
[Not working code.zip](https://github.com/MarlinFirmware/Marlin/files/5476296/Not.working.code.zip)

boelle, 2020-11-02:
Not working code.zip does not extract, please attach again and make sure its not empty zip

aghfars, 2020-11-02:
Unretract is acting as retract after every retact unless the retract is been deleted
[Not Working Gcode.zip](https://github.com/MarlinFirmware/Marlin/files/5476431/Not.Working.Gcode.zip)

boelle, 2020-11-02:
`;Prime the extruder
G92 E0
G1 F200 E5.4
G92 E0
G92 E0
G92 E0
G1 F1500 E-2
;LAYER_COUNT:99
;LAYER:0
M107
M204 S500
G0 F1200 X145.705 Y140.863 Z0.15
M204 S200
;TYPE:SKIRT
G1 F1500 E0`

the start looks very strange

you have 3 lines with G92 E0 only one is needed

`;Prime the extruder
G92 E0
G1 F200 E5.4
G92 E0
G92 E0
G92 E0
;LAYER_COUNT:99
;LAYER:0
M107
M204 S500
G0 F1200 X145.705 Y140.863 Z0.15
M204 S200
;TYPE:SKIRT
G1 F1500 E0`

boelle, 2020-11-02:
so how i would read this stump of code:

- reset extrusion count
- prime extruder with 5.4mm extrusion
- reset extrusion count
- reset extrusion count
- reset extrusion count
- retract 2mm
- switch off all fans
- set acc. to 500
- move to X145.705 Y140.863 Z0.15
- set acc to 200
- unretract 2mm (becaused it is retracted 2mm)

boelle, 2020-11-02:
this is my start gcode, i only reset extrusion count 2 times, once before prime and one after prime just before print starts. i do not do a retract as i do not have a problem with ooze

- M420 S1; Load bed level mesh
- G28 Z F750; Home Z
- G28 Y F750; Home Y
- G28 X F750; Home X
- G92 E0 ; zero the extruded length
- G1 X50 Y0 F750 ; go to the corner of the front
- G1 Z0.1 ; get close to the bed
- M190 S[first_layer_bed_temperature] ; wait for bed temp
- M109 S[first_layer_temperature] ; wait for extruder temp
- G1 X100 E10 F200 ; make a thick line to prime extruder
- G92 E0 ; zero the extruded length
- G1 Z0.3 ; lift Z
- G4 S2   ;wait 2 sec to bleed off excess pressure

aghfars, 2020-11-02:
> so how i would read this stump of code:
>
> * reset extrusion count
> * prime extruder with 5.4mm extrusion
> * reset extrusion count
> * reset extrusion count
> * reset extrusion count
> * retract 2mm
> * switch [...]

yes but Actually the unretract after retract acting as retract again
and retract at the start of gcode is not the only one in the gcode

boelle, 2020-11-02:
i looked at your configs

/*
======================================================================================================================================================
edited files:
platformio.ini
Marlin\Configuration.h done
Marlin\Configuration_adv.h done
Marlin\_Bootscreen.h done
Marlin\_Statusscreen.h done
Marlin\pins\ramps\pins_RAMPS.h done
Marlin\src\lcd\dogm\dogm_Bootscreen.h done
Marlin\src\inc\Conditionals_LCD.h done
Marlin\src\inc\Version.h done
Marlin\src\core\language.h done
======================================================================================================================================================
*/

is that added by you? should normally not be there

boelle, 2020-11-02:
i'm not expert so someone else will have to chime in @ellensp @rhapsodyv 

but i would suggest you use bugfix 2.0.x and only change configuration.h, configuration_adv.h and platformio.ini

just to rule things out

aghfars, 2020-11-02:
> i looked at your configs
>
> # /*
> # edited files:
> platformio.ini
> Marlin\Configuration.h done
> Marlin\Configuration_adv.h done
> Marlin_Bootscreen.h done
> Marlin_Statusscreen.h done [...]

yes i add this because of my handmade OLED and another text or something like that and Every time a new version comes, I do not forget which parts I have to change

boelle, 2020-11-02:
have you tried bugfix?

aghfars, 2020-11-02:
> i'm not expert so someone else will have to chime in @ellensp @rhapsodyv
> 
> but i would suggest you use bugfix 2.0.x and only change configuration.h, configuration_adv.h and platformio.ini
> 
> just to rule things out

I can test in three days because I have an exam

ellensp, 2020-11-06:
You need to attach all files you edited.. I would suspect a pin conflict, something else is conflicting the extruder direction pin.

aghfars, 2020-11-06:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5499177/Marlin.zip)
I put all the files
I found the problem
This problem occurs when I enable the LIN_ADV feature even on bugfix.

boelle, 2020-11-06:
if you truely mean the issue is fixed then click the green close button below

aghfars, 2020-11-06:
> if you truely mean the issue is fixed then click the green close button below

i want to use LIN_ADV What should I do?

sjasonsmith, 2020-11-07:
It sounds like you might seeing one retract from your G-code, and also the "Linear reverse" after an E movement with Linear Advance enabled.

Linear Advance works by advancing extra filament when starting E movement. It pulls back that extra filament when the extruder slows down. I suspect you are seeing this, followed by the -2 retract in your GCode.

boelle, 2020-11-08:
> i want to use LIN_ADV What should I do?

enable it, and as @sjasonsmith said you will see extra retract, that is not a bug but LIN_ADV working its magic

sjasonsmith, 2020-11-08:
Based on previous comments, I don't think there is actually an issue here and will go ahead and close this.

closed 2020-11-08 by sjasonsmith