MarlinFirmware/Marlin #19983: [BUG] G29 (ABL) leads to M112 (Error: Probing Failed)

gregor2005, 2020-11-01:
### Bug Description

I added a PINDA probe to my Prusa i3 with RAMPS 1.4 and Marlin 2.0.7. G28 works like expected but G29 leads to problem you see below.

### Configuration Files

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/5471008/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/5471009/Configuration_adv.h.txt)

### Steps to Reproduce

Codes send via OctoPrint Terminal

1. G28
Send: G28
Recv: X:95.00 Y:87.00 Z:0.00 E:0.00 Count X:7600 Y:6960 Z:0
Recv: ok
Send: M113 S2
Recv: ok

2. G29
Send: G29
Recv: echo:Unknown command: ""
Recv: ok

3. G29
Send: G29
Recv: Error:Probing Failed
Changing monitoring state from "Operational" to "Error: Probing Failed"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error: Probing Failed" to "Offline (Error: Probing Failed)"
Connection closed, closing down monitor

**Expected behavior:**

Auto bed leveling

**Actual behavior:**

Printer stopped with M112 (Error: Probing Failed)

boelle, 2020-11-01:
what if you send M119 ?

gregor2005, 2020-11-03:
Hi, I found the Problem.
When debugging is turned on I figured out that it triggers because it searches the probe on the wrong pin, I had to add the following parameter to the configuration file:

    #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

closed 2020-11-03 by gregor2005