MarlinFirmware/Marlin #19984: [FR] Automatically disable HOTEND_IDLE_PROTECTION after printing initiated. Automatically pause printing if PREVENT_COLD_EXTRUSION is triggered

Drzet, 2020-11-01:
### Description

It seems like HOTEND_IDLE_PROTECTION timer can run in parallel (and disregarding) with M109 command. On several occasions I had hotend disabled almost right after it reaches printing temperature. Then, while printer performs G28 and G29 J commands, it cools enough to prevent extrusion.
Also, in my dual extruder system, the inactive hotend gets deactivated while idle and even after reaching PREVENT_COLD_EXTRUSION threshold printing continues - with anything being extruded.

### Feature Workflow

1. Preheat hotend
2. [Second Action]
3. [and so on...]

#### Additional Information
It obviously can be circumvented by simply making idle protection interval longer, but while in dual extrusion mode inactive extruder can be idle for unpredictable amount of time, depending on model/layer. Therefore I think idle protection should be completely inactive while printing.
In case of cold extrusion prevention kicking in printing should be automatically paused, or, even better, hotend automatically re-heated.

legend069, 2020-11-05:
i think i've narrowed it down to if i preheat the bed/nozzle and then start a print the idle time out occurs.
works fine if i don't preheat it

(preheating from print menu, not start gcode )

Drzet, 2020-11-05:
> i think i've narrowed it down to if i preheat the bed/nozzle and then start a print the idle time out occurs.
> works fine if i don't preheat it
> 
> (preheating from print menu, not start gcode )

Perhaps, but nevertheless idle timer should stop once printing commences

rhapsodyv, 2020-11-22:
We can't disable HOTEND_IDLE_PROTECTION while printing. In fact, is intend to protect a print interruption that marlin could not about by itself.
For example: you are printing from a host, and the host just stop sending commands in the middle of the print. HOTEND_IDLE_PROTECTION is what will protect your printer.

You can tune it to better fit your needs, or disable at all. You have all the options.

thisiskeithb, 2021-09-24:
`HOTEND_IDLE_TIMEOUT`, formally `HOTEND_IDLE_PROTECTION`, will no longer trigger early: #21535

closed 2021-09-24 by thisiskeithb