MarlinFirmware/Marlin #19988: [BUG] babystepping goes on X axis instead of Z on coreXZ machine

exec79, 2020-11-01:
### Bug Description

everything works fine regarding of directions on a coreXZ machine except babystepping. it makes correction moves on X axis insstead of Z.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

 using doubleclick to get babystep correction, 

1. start printing
2. using doubleclick to get babystep correction
3. adjusting babysteps

**Expected behavior:** move Z the amount i tune with babystepping

**Actual behavior:**X moving the correction steps instead of Z.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5471674/Marlin.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

exec79, 2020-11-01:
forgot to add, its 2.0.7.2 release

boelle, 2020-11-02:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

exec79, 2020-11-02:
i did setup only essential settings but still this (seems) bug exist. during printing, babysteps moves in X direction not Z. One of the motors should move the opposite direction i guess. Thank you!
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5477132/Marlin.zip)

mchlbatista, 2020-11-05:
I can confirm the bug. Getting same issue on CoreXZ Marlin Bugfix2.0.7. SKR 1.4 TMC2209.
Babystepping is affecting X axis position and not Z axis position.

mchlbatista, 2020-11-10:
It's happening when using `probe z-offset`. `During printing` is not a condition here, it's always happening.

zipeartnes, 2022-01-09:
Is there a solution to this??

mchlbatista, 2022-07-07:
> Same issue here, with 2.0.9.3.x version. Is there a solution??

Same here… started using Klipper. Ton of bugs went away.

diegostf, 2022-07-08:
Using klipper, does the HMI still work (I have a BiQU B1)? Is it possible to use for settings/status?

thinkyhead, 2023-04-29:
Anyone have a CoreXZ machine who would be able to help delve into the issue? It's probably a very simple fix. The babystepping code in general is not complicated, so it would be easy to experiment and even accidentally hit on a solution. So, let's give it a shot - wot say?

thinkyhead, 2023-04-29:
For instance, we might try changing this line…
```cpp
BABYSTEP_CORE(X, Z, direction, BABYSTEP_INVERT_Z, (CORESIGN(1)<0));
```
…to this…
```cpp
BABYSTEP_CORE(Z, X, direction, BABYSTEP_INVERT_Z, (CORESIGN(1)<0));
```
…just to see what happens.

And maybe it would also require changing `(CORESIGN(1)<0)` to `(CORESIGN(1)>0)`… so we could try that too.

thinkyhead, 2023-04-29:
Assuming CoreXY provides an isomorphic model, and assuming that XY babystepping is sorted out on that arrangement, we can make a logical comparison.

X versus Y babystepping on CoreXY:
```cpp
BABYSTEP_CORE(X, Y, direction, 0, 0);  // X Babystepping
BABYSTEP_CORE(X, Y, !direction, 1, (CORESIGN(1)>0)); // Y Babystepping
```
X babystepping on CoreXZ:
```cpp
BABYSTEP_CORE(X, Z, direction, 0, 0);
```
… so maybe Z babystepping could be…?
```cpp
BABYSTEP_CORE(X, Z, !direction, !BABYSTEP_INVERT_Z, (CORESIGN(1)>0));
```
…or…?
```cpp
BABYSTEP_CORE(X, Z, !direction, BABYSTEP_INVERT_Z, (CORESIGN(1)>0));
```

Core kinematics hurts my brain too much to be sure, but there are only a small number of combinations that it could be.

thinkyhead, 2023-04-29:
Ref: #18342

thisiskeithb, 2023-05-08:
Here's a copy of the original configs ported to current `bugfix-2.1.x` (d36f43d): [19988-exec79-bugfix-2.1.x-d36f43d.zip](https://github.com/MarlinFirmware/Marlin/files/11416288/19988-exec79-bugfix-2.1.x-d36f43d.zip)

closed 2023-05-08 by thinkyhead

thisiskeithb, 2023-05-08:
I plan to build an Ender-3-based Switchwire at some point, so I figured this would be a good issue to fix before hand 😄 

I set up a couple motors on my test bench and built this chart help understand how the X (left) & Z (right) motors work together:

| | X Motor (left) | Z Motor (right) |
|-:|:-:|:-:|
| Z+ | CW | CCW |
| Z- | CCW | CW |
| X- | CCW | CCW |
| X+ | CW | CW |

Using OP's config, babystepping Z down/negative actually moves the hotend in the X+ direction:

- X Motor: CW
- Z Motor: CW

This was already confirmed, but I wanted to make sure it was still an issue.

I chatted with thinkyhead on discord and reset `BABYSTEP_INVERT_Z` to `false` & used the following change in `stepper.cpp` instead of suggestions above:
```diff
diff --git a/Marlin/src/module/stepper.cpp b/Marlin/src/module/stepper.cpp
index 01a562d..4cb44af 100644
--- a/Marlin/src/module/stepper.cpp
+++ b/Marlin/src/module/stepper.cpp
@@ -3683,7 +3683,7 @@ void Stepper::report_positions() {
       case Z_AXIS: {
 
         #if CORE_IS_XZ
-          BABYSTEP_CORE(X, Z, direction, BABYSTEP_INVERT_Z, (CORESIGN(1)<0));
+          BABYSTEP_CORE(X, Z, direction, BABYSTEP_INVERT_Z, (CORESIGN(1)>0));
         #elif CORE_IS_YZ
           BABYSTEP_CORE(Y, Z, direction, BABYSTEP_INVERT_Z, (CORESIGN(1)<0));
         #elif DISABLED(DELTA)
```
Based on the following motor directions, babystepping Z should now work correctly:
- X Motor: CCW
- Z Motor: CW

Mind testing this change @exec79? Here's a copy of your original configs ported to current `bugfix-2.1.x` (d36f43d) with `BABYSTEP_INVERT_Z` set to back to default/`false`.

thinkyhead, 2023-05-08:
> Based on the following motor directions, babystepping Z should now work correctly.

Thanks for taking the initiative on This Old Lodge. Given the simplicity of the solution I would class this bug as being a minor typo. The patch is already pushed, along with a tweak to CoreXY to make it consistent with the overall babystep pattern. Thanks again to everyone who pointed at this bug on their way to the dance.

Chris-Jayden, 2023-05-24:
> I plan to build an Ender-3-based Switchwire at some point, so I figured this would be a good issue to fix before hand 😄
> 
> I set up a couple motors on my test bench and built this chart help understand how the X (left) & Z (right) motors work together:
> 
> X Motor (left)	Z Motor (right)
> Z+	CW	CCW
> Z-	CCW	CW
> X-	CCW	CCW
> X+	CW	CW
> Using OP's config, babystepping Z down/negative actually moves the hotend in the X+ direction:
> 
> * X Motor: CW
> * Z Motor: CW
> 
> This was already confirmed, but I wanted to make sure it was still an issue.
> 
> I chatted with thinkyhead on discord and reset `BABYSTEP_INVERT_Z` to `false` & used the following change in `stepper.cpp` instead of suggestions above:
> 
> ```diff
> diff --git a/Marlin/src/module/stepper.cpp b/Marlin/src/module/stepper.cpp
> index 01a562d..4cb44af 100644
> --- a/Marlin/src/module/stepper.cpp
> +++ b/Marlin/src/module/stepper.cpp
> @@ -3683,7 +3683,7 @@ void Stepper::report_positions() {
>        case Z_AXIS: {
>  
>          #if CORE_IS_XZ
> -          BABYSTEP_CORE(X, Z, direction, BABYSTEP_INVERT_Z, (CORESIGN(1)<0));
> +          BABYSTEP_CORE(X, Z, direction, BABYSTEP_INVERT_Z, (CORESIGN(1)>0));
>          #elif CORE_IS_YZ
>            BABYSTEP_CORE(Y, Z, direction, BABYSTEP_INVERT_Z, (CORESIGN(1)<0));
>          #elif DISABLED(DELTA)
> ```
> 
> Based on the following motor directions, babystepping Z should now work correctly:
> 
> * X Motor: CCW
> * Z Motor: CW
> 
> Mind testing this change @exec79? Here's a copy of your original configs ported to current `bugfix-2.1.x` ([d36f43d](https://github.com/MarlinFirmware/Marlin/commit/d36f43d80bac5ca8086bc3ca70eb0c5330f317f2)) with `BABYSTEP_INVERT_Z` set to back to default/`false`: [19988-exec79-bugfix-2.1.x-d36f43d.zip](https://github.com/MarlinFirmware/Marlin/files/11417284/19988-exec79-bugfix-2.1.x-d36f43d.zip).
> 
> Edit: This fix was already pushed to bugfix ([f207177](https://github.com/MarlinFirmware/Marlin/commit/f2071770984e393ed524d924ff501fc2cd42849d)), so you don't need to make the change to `stepper.cpp`. Just drop the updated configs into current bugfix and build away.

Nice, work!~
I just built my Ender 3 (v2) Switchwire Rev 2.0 as of 2 weeks ago, so this fix will definitely be a worth addition to the firmware!~
I've been using UBL with my Voron Stealthburner UnTap and have been getting perfect first layers except when I forget to heat and wipe/clean the nozzle beforehand; this bugfix will help me quick-fix if the situation arises again in the future!~

Only 1 more feature left before I'd consider my Ender 3 (v2) Switchwire Rev 2.0 as perfect - Z-Axis Input shaping for  the Core XZ kinematics~

commit 7cf10657 (2023-05-08): 🐛 Fix Babystepping for CoreXZ

Fixes #19988
commit f2071770 (2023-05-08): 🐛 Fix Babystepping for CoreXZ

Fixes #19988