MarlinFirmware/Marlin #19993: add support for the m355 T gcode to switch the printer's case backlight with a single command

ssavickiy, 2020-11-02:
to the GcodeSuite::M355() function Marlin\src\gcode\feature\caselight\M355.cpp
add the following lines before the _if (args) update_case_light();_
line
_if (parser.seenval('T')) {
++args, case_light_on = !case_light_on;
case_light_rg_flag = true;
}_

then the code M355 T will trigger on/off illumination. It is convenient to make an octoprint in the web interface with one button.

sjasonsmith, 2020-11-07:
It sounds simple enough. That doesn't appear to be an argument already used for this purpose on the [RepRap Wiki](https://reprap.org/wiki/G-code#M355:_Turn_case_lights_on.2Foff).

Your best option is to prepare a pull request yourself with the proposal, especially since you have already figured out how to implement it.

ssavickiy, 2021-01-16:
i dont know)