MarlinFirmware/Marlin #19994: [BUG] Using Bugfix-2.0.x branch, SKR PRO V1.1 with TMC2209 on X, Y, Z and E0 in UART Mode & Adafruit MAX31865 board in Hardware SPI mode on EXP2 contanstaly shows "Temp measurement Errors" even tho PT100 Sensor is showing the correct Temperature

GadgetAngel, 2020-11-02:
### Bug Description

Using bugfix-2.0.x branch from today, with SKR PRO V1.1 board which has BIQU TMC2209 on X, Y, Z and E0 in UART Mode and Adafruit MAX31865 board in Hardware SPI mode on EXP2 with a TFT25 V2.0 touch screen on TFT port.  The PT100 sensor reads the correct temperature but Marlin continuously produces "Temp measurement errors".  This Temp measurement errors does not make any sense and should not be produced.

### Configuration Files

Configuration.h file and Configuration_adv.h file and pins_BTT_SKR_PRO_common.h file:
[Configuration.h & Configuration_adv.h & pins_BTT_SKR_PRO_common.h files.zip](https://github.com/MarlinFirmware/Marlin/files/5472068/Configuration.h.Configuration_adv.h.pins_BTT_SKR_PRO_common.h.files.zip)

The shared link from my google drive for the whole Marlin directory in .zipped file with platformio.ini file included:

https://drive.google.com/file/d/1cpCbs3MlYXtWdZFSMAbJFUegMu4PzkTp/view?usp=sharing

### Steps to Reproduce

1. Download the latest bugfix-2.0.x branch of Marlin
2. Hook up a Adafruit MAX31865 board as shown in the diagram below:

[image]

3. set pins_BTT_SKP_PRO_common.h file as below:
[image]

4. set in **configuration.h** the following:
Set SERIAL_PORT to -1
Enable SERIAL_PORT_2 and set to 1
Change the Marlin variable TEMP_SENSOR_0 to -5
Ensure the Marlin variable EXTRUDERS is set to 1
Change HEATER_0_MAXTEMP to 350
Ensure the Marlin variable SDSUPPORT is enabled.
Enable Marlin variable MAX31865_SENSOR_OHMS and is set to 100
Enable Marlin variable MAX31865_CALIBRATION_OHMS and is set to 430

4. ( 4. continued ) In **platformio.ini** ensure the following (do not have a later version then 1.1.0 of Adafruit MAX31865 library installed): 
MAX6675_IS_MAX31865 = Adafruit MAX31865 library@~1.1.0

4. ( 4. continued ) set in **configuration_adv.h** file the following: 
Ensure Marlin variable THERMOCOUPLE_MAX_ERRORS is enabled and set to 20
Ensure Marlin variable SHOW_TEMP_ADC_VALUES is enabled
Enable MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED and set to 10

Wire the Adafruit MAX31865 board to the SKR PRO V1.1 as shown in the above diagram.
Compile and perform a M502 followed by M500 to ensure the new Marlin setup is used by the SKR PRO V1.1 board.
Turn on the printer and hook up pronterface to the SKR PRO V1.1 board
If you look at the TFT screen you will see it takes about 30 second for the TFT screen to show connected to the printer but the TFT screen will show the temperature of the PT100 sensor after 30 seconds.  When you connect pronterface to the SKR PRO V1.1 board after this, you immediately see "Temp measurement error! MAX6675" but I can place the PT100 sensor close to a soldering iron and see the temperature change reflected on the TFT screen on in pronterface.  Even tho I am getting good temperature readings pronterface continues to put out "Temp measurement error! MAX6675".  **If a users sets up there SERIAL_PORT to 1 then the TFT screen will disconnect due to the "Temp measurement error! MAX6675" being produced by Marlin.**
 
**Expected behavior:**

I expect Marlin not to be putting out "Temp measurement error! MAX6675" message all the time, while still delivering a good temperature reading from PT100 sensor.

**Actual behavior:**

Turn on the printer and hook up pronterface to the SKR PRO V1.1 board
If you look at the TFT screen you will see it takes about 30 second for the TFT screen to show connected to the printer but the TFT screen will show the temperature of the PT100 sensor after 30 seconds.  When you connect pronterface to the SKR PRO V1.1 board after this, you immediately see "Temp measurement error! MAX6675" but I can place the PT100 sensor close to a soldering iron and see the temperature change reflected on the TFT screen on in pronterface.  Even tho I am getting good temperature readings pronterface continues to put out "Temp measurement error! MAX6675".  **If a users sets up there SERIAL_PORT to 1 then the TFT screen will disconnect due to the "Temp measurement error! MAX6675" being produced by Marlin.**

#### Additional Information

Here is a video of the bootup process, you can see I never touch the TFT screen and the TFT screen will after about 30 seconds start displaying the temperature for the PT100 sensor:

https://drive.google.com/file/d/1OGdkTxWqfj61Y-6039-VQrJ3G60MO_hF/view?usp=sharing

I checked my PT100 sensor with an Ohm meter and it reads approximately 110 Ohm.

I connected pronterface (http://kliment.kapsi.fi/printrun/) to my board via the USB connection and found I was getting a lot of "Temp measurement errors" but the SKR PRO V1.1 board was still able to read the PT100 sensor temperature values. Here is what the errors look like:

[image]

Here is are pictures of my hook up:

Green wire is CS of Adafruit MAX31865 board
Purple wire is SDI of Adafruit MAX31865 board
Brown wire is SDO of Adafruit MAX31865 board
White wire is CLK of Adafruit MAX31865 board
Black wire is GND of Adafruit MAX31865 board
Red wire is VIN of Adafruit MAX31865 board

[image]

[image]

[image]

[image]

[image]

Here is the output from the compile:

[image]

---

sjasonsmith, 2020-11-07:
A lot of users have mentioned MAX31865 lately, but this is the first I've seen this reported. Maybe one of them can help verify whether they see the same behavior. I don't have the hardware I would need to try to reproduce it.

GadgetAngel, 2020-11-08:
> A lot of users have mentioned MAX31865 lately, but this is the first I've seen this reported. Maybe one of them can help verify whether they see the same behavior. I don't have the hardware I would need to try to reproduce it.

Please see this from @bob808.  He saw it way back in Nov 27,2019 (https://github.com/MarlinFirmware/Marlin/issues/5723#issuecomment-558941951) I am just verifying what he saw.  I will be submitting a Pull Request in a couple of days that fixes this problem.   

On top of "temp error" bug, I found another bug with hardware SPI:  the second Adafruit MAX31865 board was not showing a correct temperature because Marlin never instantiated the second Adafruit MAX31865 board on TEMP_SENSOR_1.  

The way Marlin works right now you can only use ONE Adafruit MAX31865 board.  

Again my Pull Request fixes this issue.  I also allow the user to define a PT100 on E0 and PT1000 on E1  (or a PT1000 on E0 and PT100 on E1) since they both use the Adafruit MAX31865 library calls.  At present you can only define a PT100 on E0 and/or E1 or and PT100 on E0 and/or E1.

To fix the "temp error" bug I added code to setup the MAX6675 bit mask correctly based on the MAX31865 datasheet (https://datasheets.maximintegrated.com/en/ds/MAX31865.pdf).  I also do a call to the Adafruit MAX31865 library to poll the device for faults by using object.readFault();.  If the number return is greater than 0 than a fault occurred and I indicate which of the seven faults was register then clear the fault and move on.

After doing this it cleared away the "temp error" messages. With out this fix all the error message produced were making users which use a TFT screens (as the primary serial port) impossible to use.  Basically the Adafruit MAX31865 was using the MAX6675 board's bit mask because it was falling thru the ELSE part of the statement and the last MAX6675 device listed was the MAX6675 board.  

I am presently checking to see if two Adafruit MAX31865 boards will run in software SPI mode.  If I find problems, I plan on fixing them and submitting all of this in one Pull Request.  

But my hope is that the second instantiation of the Adafruit MAX31865 on TEMP_SENSOR_1 will fix any temperature display problems I might have found.  It all depends on the flow thru the code.  I have not yet looked at the Software SPI end of the code which uses the MAX6675_SEPARATE_SPI bus.

sjasonsmith, 2020-11-08:
It seems you have thing under control! Thanks for working to make this all work better. I look forward to seeing your contribution!

I recommend you reach out to @rhapsodyv  if you intend to rework the SPI communication. He has been working on refactoring our SPI mechanisms to be more consistent across Marlin. There is of course less control when things are done inside libraries, but he may have ideas that can help improve some of the current limitations or hackiness. We are both pretty often around on the Discord channel.

GadgetAngel, 2020-11-08:
> It seems you have thing under control! Thanks for working to make this all work better. I look forward to seeing your contribution!
>
> I recommend you reach out to @rhapsodyv if you intend to rework [...]

Well I tested the Software SPI for 2 Adafruit MAX31865 boards and it works without issue.  I am not reworking the SPI communications, I am not that smart.  I just wanted to get Marlin to be able to use two MAX31865 board due to the fact that sooo many boards are now going 32-bit and are limited to 3.3VDC.  Having the Adafruit MAX31865 work on both E0 and E1 and using SPI mode takes care of all the problems and concerns over frying the MCU on the board because SPI uses only 3.3VDC logic to talk with.  So yes, am am very very happy to help out and help fix this issue.  I hope the work I did will help Marlin.  I love the code base and it says a lot about "how good the code is" if I can figure it out.  

One question, I am assuming I need to fork the bugfix branch in github desktop make all the code changes in the fork and submit the fork for PR or can I just Zip up the folder I was working in (I started with the bug fix branch published yesterday) and submit that?

sjasonsmith, 2020-11-08:
You’ll need to create your own Marlin fork in GitHub. It’s best not to modify the bugfix branch directly. Create a new branch from that to do your work, and post your pull request. I believe there is a tutorial on marlinfw.org, or head over to Discord and ask questions about the process.

GadgetAngel, 2020-11-12:
This issue can now be closed because the latest bugfix-2.0.x branch resolves the problems

closed 2020-11-12 by GadgetAngel