MarlinFirmware/Marlin #19995: pause print if tempreture drops below threshold

legend069, 2020-11-02:
### Description

Hi
so in relation to the 
#define PREVENT_COLD_EXTRUSION
#define EXTRUDE_MINTEMP 170

(this is a separate thing not trying to resolve here )

once the print starts and is running the g code to move the x,y,z,e stepper motors 
if there the EXTRUDE_MINTEMP drops below 170 ( has happened to me twice)

the extruder will stop thats great but the x,y,z keeps trying to print

Scenario;
print starts bed/hotend set

starts moving x,y,x,e steppers to make the print
something werid thing happens and the hotend/bed both drop to 0
once the temperature drops below EXTRUDE_MINTEMP PREVENT_COLD_EXTRUSION gets triggered. thats great
 x,y,z steppers keep moving to try make the print but its no long extruding

in theory you would want the print to be paused (if you're 1 hour+ into a print )

but the only way i think this can be done is to add another variable in the firmware and that the slicer makes that
will enable/disable stepper motor movement.

obviously you still want to move the steppers when cold for various reasons
but need them to move when it's actually printing

### Feature Workflow

variable actuallyPinting ; added by start gcode from slicer

check state of actuallyPinting ;
while (actuallyPinting == true){
gcode starts to print
continusly checks actuallyPinting state to stay in loop for control of x,y,z steppers
}
if(actuallyPinting == false){
pause print wait for user to resume 
}

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

sjasonsmith, 2020-11-02:
> if there the EXTRUDE_MINTEMP drops below 170 ( has happened to me twice)

Shouldn't the inability to keep the hotend at temperature have triggered thermal protection?

legend069, 2020-11-02:
> 
> 
> > if there the EXTRUDE_MINTEMP drops below 170 ( has happened to me twice)
> 
> Shouldn't the inability to keep the hotend at temperature have triggered thermal protection?

when it happened to me the first time i had to pause it and after it reheated it purged and resumed printing.

if it was thermal protection that got triggered the print would of paused and told me to reset it correct ?

hamster65, 2020-11-02:
No, thermal protection acts as an emergency stop. You have to reset the printer afterwards.
In normal operation. thermal protection should never trigger. It is to protect your printer from causing fires.
If you have issues with temperature drops, than something is wrong with your hardware our your PID regulator calibration.

Drzet, 2020-11-02:
https://github.com/MarlinFirmware/Marlin/issues/19984#issue-733979346

legend069, 2020-11-03:
> No, thermal protection acts as an emergency stop. You have to reset the printer afterwards.
> In normal operation. thermal protection should never trigger. It is to protect your printer from causing [...]

yes so it wasn't thermal protection that got triggered. something else caused it.

anyhow is it a decent feature ?

legend069, 2020-11-03:
> 
> 
> [#19984 (comment)](https://github.com/MarlinFirmware/Marlin/issues/19984#issue-733979346)

yes i think that may of happened to me.
upvote