MarlinFirmware/Marlin #19996: [BUG] Sensorless homing on CoreXY might false trigger on second homing axis

soimy, 2020-11-02:
### Bug Description

The nature of corexy setup is dual stepper movement on a single axis. On the sensorless homing process after the first axis homed, the blocker of the first axis will always apply torque to both steppers while homing the second axis, thus sometimes false trigger the endstop signal of TMC stepper driver.

If `STALL_SENSITIVITY` is set to a low value, the second axis is more likely to false homing after a successful first axis homing.

### Misc info
None, only theory explained.

boelle, 2020-11-02:
is this a bug report or are there a question/discussion here?

Portzal, 2020-11-11:
Are you suggesting that backoff after homing should be automatically applied for sensor less homing? I found myself that using a 5mm backoff after homing helped get my Cartesian working reliably. I applied the same settings to my coreXY with good results.

Portzal, 2020-11-11:
I use backoff after homing to achieve reliable result.

TadyTheFish, 2020-11-12:
Do you mean
`#define SENSORLESS_BACKOFF_MM  { 2, 2 }`
or
`//#define HOMING_BACKOFF_POST_MM { 2, 2, 2 }`
or both?

Portzal, 2020-11-13:
I am using the following options on my CoreXY and Cartesian.
Both are trouble free and have a reasonably wide tuning band.
Configuration.h
#define HOMING_FEEDRATE_XY (30*60)
Configuration_adv.h
#define HOMING_BUMP_MM          {  0,  0,  0  }
#define HOMING_BACKOFF_POST_MM  {  5,  5,  0  }
#define X_CURRENT_HOME  (half of full current) 
#define Y_CURRENT_HOME  (half of full current)
#define SENSORLESS_HOMING
#define X_STALL_SENSITIVITY  (tuned to machine)
#define Y_STALL_SENSITIVITY  (tuned to machine)
#define TMC_HOMEPHASE  {  896,  896  ,  -1  } (still trialling but doesn't seem to cause any issues)
I found #define QUICK_HOME caused issues with sensor less homing, if Y homed before X, Y would hold on and not back off until X homed. Leading to noisy and inconsistent homing.
would the feature request be to automatically apply #define HOMING_BACKOFF_POST_MM  {  5,  5,  0  }  when sensor less homing is enabled?

soimy, 2020-11-13:
I think 
`#define HOMING_BACKOFF_POST_MM { 2, 2, 0 }`
shall be enabled by default when `#define SENSORLESS_HOMING` enabled
At least, this shall be mentioned in comment and docs

closed 2026-08-18 by github-actions[bot]