MarlinFirmware/Marlin #19997: TMC2226 not included as sensorless homing or probing driver in Sanity Check

hackebike, 2020-11-02:
I am using TMC 2226 as drivers with sensorless homing and probing.
Tried to compiled for SKR GTR and got this error:
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/5476433/Archive.zip)

In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/HAL.cpp:28:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1307:8: error: #error "SENSORLESS_PROBING requires a TMC2130/2160/2209/5130/5160 driver on Z."
 1307 |       #error "SENSORLESS_PROBING requires a TMC2130/2160/2209/5130/5160 driver on Z."
      |        ^~~~~
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2591:6: error: #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
 2591 |     #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
      |      ^~~~~
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2612:4: error: #error "SENSORLESS_PROBING requires TMC2130, TMC2160, TMC2209, TMC2660, or TMC5160 stepper drivers."
 2612 |   #error "SENSORLESS_PROBING requires TMC2130, TMC2160, TMC2209, TMC2660, or TMC5160 stepper drivers."
      |    ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/HAL_SPI.cpp:25:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1307:8: error: #error "SENSORLESS_PROBING requires a TMC2130/2160/2209/5130/5160 driver on Z."
 1307 |       #error "SENSORLESS_PROBING requires a TMC2130/2160/2209/5130/5160 driver on Z."
      |        ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/MarlinSerial.cpp:22:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1307:8: error: #error "SENSORLESS_PROBING requires a TMC2130/2160/2209/5130/5160 driver on Z."
 1307 |       #error "SENSORLESS_PROBING requires a TMC2130/2160/2209/5130/5160 driver on Z."
      |        ^~~~~
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2591:6: error: #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
 2591 |     #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
      |      ^~~~~
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2612:4: error: #error "SENSORLESS_PROBING requires TMC2130, TMC2160, TMC2209, TMC2660, or TMC5160 stepper drivers."
 2612 |   #error "SENSORLESS_PROBING requires TMC2130, TMC2160, TMC2209, TMC2660, or TMC5160 stepper drivers."
      |    ^~~~~
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/Sd2Card_sdio_stm32duino.cpp:23:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1307:8: error: #error "SENSORLESS_PROBING requires a TMC2130/2160/2209/5130/5160 driver on Z."
 1307 |       #error "SENSORLESS_PROBING requires a TMC2130/2160/2209/5130/5160 driver on Z."
      |        ^~~~~
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2591:6: error: #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
 2591 |     #error "SENSORLESS_PROBING requires a TMC stepper driver with StallGuard on Z."
      |      ^~~~~
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2612:4: error: #error "SENSORLESS_PROBING requires TMC2130, TMC2160, TMC2209, TMC2660, or TMC5160 stepper drivers."
 2612 |   #error "SENSORLESS_PROBING requires TMC2130, TMC2160, TMC2209, TMC2660, or TMC5160 stepper drivers."

thisiskeithb, 2020-11-02:
TMC2226 is not a valid option. Use TMC2209: https://github.com/MarlinFirmware/Marlin/issues/18328#issuecomment-647557206
<hr>
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-02 by thisiskeithb

hackebike, 2020-11-02:
So TMC 2226 is not planned to be supported and it should be set as TMC 2209 as before.

thisiskeithb, 2020-11-02:
As far as Marlin is concerned, the TMC2226 is just a TMC2209 in a different package. Use `TMC2209` or `TMC2209_STANDALONE` for each `*_DRIVER_TYPE`.

I'll try to put in a PR later adding an explanation under the TMC section that TMC2225 = TMC2208 and TMC2226 = TMC2209.