MarlinFirmware/Marlin #19998: Dual nozzles issue Marlin 2.0.x bugfix with SKR 1.4

sckeylog, 2020-11-02:
### Bug Description
Im currently trying to add another nozzle, another extruder to my printer. My printer work very fine with SKR 1.4 and marlin 2.0. 
but when i start to config marlin to have 2 extruders, it have a real strange bug or issue. 

i made these changes, will attach config.h and config_adv files, but basicly : 
#define EXTRUDERS 2

#define SWITCHING_NOZZLE
#if ENABLED(SWITCHING_NOZZLE)
  #define SWITCHING_NOZZLE_SERVO_NR 0
  //#define SWITCHING_NOZZLE_E1_SERVO_NR 1          // If two servos are used, the index of the second
  #define SWITCHING_NOZZLE_SERVO_ANGLES { 0, 90 }   // Angles for E0, E1 (single servo) or lowered/raised (dual servo)
#endif

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300}

// Only power servos during movement, otherwise leave off to prevent jitter
#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

Then, i compile the code, success, upload to board, Reset board to make it load new fimware. Then bug start. LCD show up Marlin starting words, but then board keep restart over and over again... just show up the welcome word then restart. i even cant upload new code to board, restart continously. 
somehow, when i manage to upload old fimware.bin file to the board, it running normal again. 

please help me!!

#### Additional Information

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5476695/Marlin.zip)

https://streamable.com/x0yf9f

ellensp, 2020-11-03:
your running a older version of marlin 020006
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

sjasonsmith, 2020-11-07:
Bug reports are really a lot easier to read when you follow the template provided. I get tired of reading the description long before encountering any actual bug report.

shitcreek, 2020-11-17:
For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-11-17 by shitcreek