MarlinFirmware/Marlin #19999: [BUG] 3DTouch probe do not used in lates versions

Fires04, 2020-11-02:
### Bug Description

I try to update my firmware to the latest 2.0.x - bugfix, so hand merges my old configs from 2.0.6.1 to the latest version but I got a problem with the 3DTouch probe. It all looks good, test works, stow, and deploy also works. But If I send G28 the probe does not deploy and hotend try to smash to bed. even if I manually pull the probe. I try it in the 2.0.0 master branch (2.11.2020) also in 2.0.x-bugfix same error.  (the main reason for the update is the bug in M600 and octoprint that should be repaired in bugfix branch).

Board: SKR1.4 turbo
Printer Ender3Pro
Probe: 3DTouch

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5476889/Marlin.zip)

### Steps to Reproduce

1. Clone latest bug-fix branch
2. Setup config files like in the attachment
3. Setup HW
4. Flash
5. Try home all axis

**Expected behavior:**

Home X and Y work ok after that the probe should go to the middle of the bed, rise a little at Z, deploy probe, go down until probe trigger.

**Actual behavior:**

Home X and Y work ok after that the probe goes to the middle of the bedand go straight down to smash the bed without stopping.

#### Additional Information
* Try more version and branches
* I googled many hours
* Tried discord
* Tried Facebook group

SlackHacky, 2020-11-02:
User error I believe badly configured configuration.h

//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define Z_MIN_PROBE_PIN P0_10

// Force the use of the probe for Z-axis homing
//#define USE_PROBE_FOR_Z_HOMING

should read #define USE_PROBE_FOR_Z_HOMING 
(loose the //)

thisiskeithb, 2020-11-02:
As noted above, you have `USE_PROBE_FOR_Z_HOMING` commented out in your config, so you're expected to home with an endstop.

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-02 by thisiskeithb