MarlinFirmware/Marlin #20003: [Bricked printer] (Platformio update.cbd file bricked Qidi X-one 2 printer)

clinde04, 2020-11-02:
The chitu v6 board option built fine, but when I flashed it on my printer, it just went dark and has a soft fan noise.

### Configuration Files
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5478162/Configuration.zip)

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [Put update.cbd onto SD card]
2. [Put the sd card into the powered off printer]
3. [Booted up the printer]

**Expected behavior:**

It beeps a little and then stops so I can restart the printer and have marlin firmware on my printer.

**Actual behavior:**

It beeped and then just all the lights turned off and a fan just blows softly now. LCD is off.

#### Additional Information

rhapsodyv, 2020-11-02:
I don't think we support Qidi printers... All chitu work was done using TronXY printers.

clinde04, 2020-11-02:
Is there any known way that it could possibly work though?

rhapsodyv, 2020-11-02:
I guess it will only happen when a Qidi owner make marlin working on it...

clinde04, 2020-11-02:
Ok. Is there anything I could do to help with development, like check all the pins or test updates for it? (That is if you even wanted to make marlin compatible with Qidi printers)

rhapsodyv, 2020-11-02:
If marlin even had booted... But if marlin is not booting, we many need a very hard work.

clinde04, 2020-11-02:
Oh. yeah marlin did not boot. How could we fix this.

thisiskeithb, 2020-11-02:
Which LCD/TFT do these printer use? I don't see one enabled in your config.

Also, your `CONFIGURATION_H_VERSION` is `020007`, so please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

clinde04, 2020-11-02:
As for the lcd I don't now right now. And I will try that in just a sec

rhapsodyv, 2020-11-02:
try enabling: 
```
#define TFT_TRONXY_X5SA
#define TFT_COLOR_UI
#define TOUCH_SCREEN
```

thisiskeithb, 2020-11-02:
How did you confirm that Marlin didn't boot? Can you connect via usb/serial (OctoPrint/Pronterface, etc.) and check if you get any serial output?

clinde04, 2020-11-02:
ok one sec.

clinde04, 2020-11-02:
It booted up into the marlin screen now! But when I try to change the fans it does not do anything I could notice.

clinde04, 2020-11-02:
Also, octoprint is stuck at:
Offline (Error: No more candidates to test, and no working port/baudrate combination detected.)

rhapsodyv, 2020-11-02:
I think we can start with the X5SA example from Marlin: https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Tronxy/X5SA

closed 2020-11-02 by thisiskeithb