MarlinFirmware/Marlin #20007: @ellensp @boelle any idea why it doesn't work? I got thinking "what is different between 1.1.9 and 2.x?" and I remembered that they use different tmcstepper libraries. I think my 1.1.9 uses TMC2130Stepper, while 2.x uses TMCStepper. Could this stepper library cause the issues I'm having?

Koby88, 2020-11-03:
@ellensp @boelle any idea why it doesn't work? I got thinking "what is different between 1.1.9 and 2.x?" and I remembered that they use different tmcstepper libraries. I think my 1.1.9 uses TMC2130Stepper, while 2.x uses TMCStepper. Could this stepper library cause the issues I'm having?

Also, could this issue be reopened?

_Originally posted by @ikoutone in https://github.com/MarlinFirmware/Marlin/issues/17478#issuecomment-699433552_

ellensp, 2020-11-03:
Please attach your configuration files.
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

boelle, 2020-11-03:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Koby88, 2020-11-03:
I hawe tested bugfix with no success. Here is my configuration and connection pictures, sorry for dirt and dust. You can see that board is not powered it is coz i take her out so i can take some pictures. I try all kind of connections with no joy, then go back to marlin 1.9 configure all and it is working without issue i hope you can help.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5484417/Marlin.zip)
[image]
[image]

Koby88, 2020-11-03:
I change topic name coz i name it by mistake Mgs Base instead Mks Gen

ellensp, 2020-11-03:
You also need to define the following in Configuration_adv.h, or they are also setup to use pin 40 and 42 
#define Z_CS_PIN          -1
#define E0_CS_PIN         -1
and add
#define E_MUX1_PIN  -1 as it also uses 42

ellensp, 2020-11-04:
Are you using vscode or arduino ide to compile?
The reason I ask is that arduino IDE you have to manually install the library TMCStepper version 0.7.1 when vscode will download and install correct version automatically.

Koby88, 2020-11-04:
If i understand you correct i just need to take of **//** before #define Z_CS_PIN -1, #define E0_CS_PIN -1 and #define E_MUX1_PIN -1 or i need do asign other pins for them ? I use vscode

Koby88, 2020-11-05:
> Are you using vscode or arduino ide to compile?
> The reason I ask is that arduino IDE you have to manually install the library TMCStepper version 0.7.1 when vscode will download and install correct version automatically.

Thanks a lot, this worked for me :) You can close the topic now

Koby88, 2020-11-05:
> You also need to define the following in Configuration_adv.h, or they are also setup to use pin 40 and 42
> #define Z_CS_PIN -1
> #define E0_CS_PIN -1
> and add
> #define E_MUX1_PIN -1 as it also uses 42

Thanks a lot @ellensp that worked for me :) You can close the topic now

closed 2020-11-05 by thisiskeithb