MarlinFirmware/Marlin #20012: [BUG] MULTIPLE_PROBING is ignored

salacpavel, 2020-11-03:
#19851 # Bug Description

MULTIPLE_PROBING variable is ignored on 2.0.7.1 build. Instead, default number (2) is always used (manual leveling)

### Config files
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5480297/Marlin.zip)

### Steps to Reproduce

configure with #define MULTIPLE_PROBING 4, build

**Expected behavior:**

G28 Z should probe the bed 4 times

**Actual behavior:**

G28 Z probes bed 2 times, as if MULTIPLE_PROBING was still commented out

thisiskeithb, 2020-11-03:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

qwewer0, 2020-11-03:
Isn't `MULTIPLE_PROBING` for probing and not for homing? So `G28 Z` should still probe 2 times?

thisiskeithb, 2020-11-03:
> Isn't `MULTIPLE_PROBING` for probing and not for homing? So `G28 Z` should still probe 2 times?

Yes. There’s only one bump while homing.

Closing since this is just a misunderstanding of the feature and not a bug.

closed 2020-11-03 by thisiskeithb