MarlinFirmware/Marlin #20015: XY axis cannot move only after homing with MKS TMC2160_OC driver

Pikoting, 2020-11-03:
### Bug Description

XY axis cannot move after homing with MKS TMC2160_OC in stealth chop mode on MKS Gen L v2.0, axis can only move again when I disable active motor, or select in spreadcycle mode.

XY axis can move if I don't home, it work fine, issue only occur after homing.  

My 3d printer is Boxzy v1.3 (discontinued) with MKS Gen L v2.0 upgrade. Before I swap out to TMC2160_OC, I have both TMC2209 (z-axis & extruder) and TB6600 (X & Y axis) driver which work well. I only swap out TB6600 as it is quite noisy. 

I have tried with both pronterface and repetier to manually control the motion or try printing something. Both axis stay in the same position but pronterface and repetier show that it is moving in different position.

MKS TMC2160_OC: https://www.aliexpress.com/item/4000274967634.html

In the config file, the tech from MKS tells me to set the driver as A4988. I did try to set A4988, TMC2160 and TMC2160_standalone, but it didn't seems to work. 

### Configuration Files

https://drive.google.com/drive/folders/1BpmcKAtu1pXJoLY54NxszcnYr2Cv3Uwz?usp=sharing

### Steps to Reproduce

1. Press homing X & Y axis or home all
2. X & Y axis home correctly 
3. Manually enter Gcode or print object
4. XY axis cannot move
5. Press disable motor.
6. X & Y axis can move again.

**Expected behavior:**

X & Y axis should move according to the command.

**Actual behavior:**

X & Y axis did not move but coordination display as moving.

I would be appreciated if anyone can help out. Thanks!

Pikoting, 2020-11-04:
Configuration Files:
[marlin config.zip](https://github.com/MarlinFirmware/Marlin/files/5486899/marlin.config.zip)

ellensp, 2020-11-04:
It sound like you have the enable inverted from what the driver expects.
So you have to disable it to get movement

actual useful information about this driver here https://github.com/makerbase-mks/MKS-Big-Current-Driver/tree/master/MKS%20TMC2160-OC 

Circuit diagram and datasheet also read that enable is high on this device, vs low of most drivers.

try This in Configuration.h
#define X_ENABLE_ON 1
#define Y_ENABLE_ON 1

Pikoting, 2020-11-04:
> It sound like you have the enable inverted from what the driver expects.
> So you have to disable it to get movement
>
> actual useful information about this driver here [...]

Thanks I will change it and see if it works.

Pikoting, 2020-11-04:
> It sound like you have the enable inverted from what the driver expects.
> So you have to disable it to get movement
>
> actual useful information about this driver here [...]

So seems to be no luck with that. It could not move entirely in both spreadcycle and stealth chop mode. Thanks anyway for the suggustion.

teemuatlut, 2020-11-07:
Post your `M122 XY` and `M122 XYV` when you think the drivers should be enabled and disabled. I need to see the IOIN register to know what the driver itself sees in each case.

Pikoting, 2020-11-08:
> Post your `M122 XY` and `M122 XYV` when you think the drivers should be enabled and disabled. I need to see the IOIN register to know what the driver itself sees in each case.

18:03:45.559 : M122 XY
18:03:45.975 : 
18:03:45.975 : Address
18:03:45.975 : Enabled
18:03:45.975 : Set current
18:03:45.975 : RMS current
18:03:45.975 : MAX current
18:03:45.975 : Run current
18:03:45.975 : Hold current
18:03:45.975 : CS actual
18:03:45.975 : PWM scale
18:03:45.975 : vsense
18:03:45.975 : stealthChop
18:03:45.975 : msteps
18:03:45.975 : interp
18:03:45.975 : tstep
18:03:45.975 : PWM thresh.
18:03:45.975 : [mm/s]
18:03:45.975 : OT prewarn
18:03:45.975 : pwm scale sum
18:03:45.975 : pwm scale auto
18:03:45.975 : pwm offset auto
18:03:45.975 : pwm grad auto
18:03:45.975 : off time
18:03:45.975 : blank time
18:03:45.975 : hysteresis
18:03:45.975 : -end
18:03:45.975 : -start
18:03:45.975 : Stallguard thrs
18:03:45.975 : uStep count
18:03:45.975 : DRVSTATUS
18:03:45.975 : sg_result
18:03:45.975 : stst
18:03:45.975 : olb
18:03:45.975 : ola
18:03:45.975 : s2gb
18:03:45.975 : s2ga
18:03:45.975 : otpw
18:03:45.975 : ot
18:03:45.975 : 157C
18:03:45.975 : 150C
18:03:45.975 : 143C
18:03:45.975 : 120C
18:03:45.975 : s2vsa
18:03:45.975 : s2vsb
18:03:45.975 : Driver registers:
18:03:45.975 : ok

Is it this one? Thank you !
As their tech from MKS, he tested the way I config and seems to have no issue at all. He is using Marlin 2.0.1 and mine is 2.0.8, maybe I should try his version and see if the problem gone once I got the time.

teemuatlut, 2020-11-08:
No that doesn't contain any of the values.

Pikoting, 2020-11-10:
> No that doesn't contain any of the values.

The drivers are connected externally like TB6600, and it doesn't not have SPI to hookup. I figure that is why it is empty. I did try the version that given by MKS tech and no luck. 

I hate to say it but I am now giving up on stealth chop mode, and use spreadcycle for now.
Thanks for the help!

closed 2020-12-18 by github-actions[bot]