MarlinFirmware/Marlin #20016: [BUG] Hotend temp reading negative

clinde04, 2020-11-03:
The hotend temp is reading negative, however the numbers are correct, there is just a "-" before the number and it tells me MINTEMP error.

### Configuration Files
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5482645/config.zip)

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

boelle, 2020-11-03:
checked that everything is connected correctly? what thermistor do you have?

clinde04, 2020-11-03:
Everything is connected properly. As for what thermistor, no clue. I have the qidi x-one2 printer and I can;t seem to figure out what thermistor it has. Everything else is working fine though.

boelle, 2020-11-03:
i know what if you connect a thermocouple wrong it shows - in front of temp
but if its a china printer i bet its just the cheapest stuff that works at shipping time

clinde04, 2020-11-03:
I can try reversing the thermocouple if that would work.

boelle, 2020-11-03:
will only work if it is a thermocouple, a thermistor is just a resistor and it will not make a difference if you reverse it

clinde04, 2020-11-03:
How would I tell the difference

clinde04, 2020-11-03:
I just looked at the cables and it looks like there are two wires that look like they're thermocouples.

clinde04, 2020-11-03:
just switched them, it did nothing. Any suggestions

closed 2020-11-03 by clinde04