MarlinFirmware/Marlin #20023: [BUG] Selecting Mixer in Menu (COLOR_UI) leads to reboot

Masanetz, 2020-11-04:
### Bug Description

Selecting Mixer in Menu leads to reboot of printer when compiled with TFT_COLOR_UI. Same config but TFT_CLASSIC_UI works flawless.

### Configuration Files

If you've made any other modifications describe them in detail here.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5488047/Configuration.zip)

### Steps to Reproduce

1. Compile bugfix-2.0.x with attached configs (TFT_COLOR_UI enabled)
2. Start printer and select "Mixer" in menu
3. --> hard reboot of printer
4. Compile with TFT_CLASSIC_UI instead of TFT_COLOR_UI
5. Start printer and select "Mixing" in menu

**Expected behavior:**

Mixer menu should be useable with COLOR_UI

**Actual behavior:**

Printer reboots hard.

#### Additional Information

Printer: Tronxy X5SA-2E with Bigtree 2in1 mixing hotend.
Mainboard: Tronxy CHITU V6
Tested with 2.0.7.2 and bugfix-2.0.x (20201104) and older...

thisiskeithb, 2020-11-04:
Tagging @rhapsodyv to look into this as requested via Discord 🙂

rhapsodyv, 2020-12-25:
Can you test this? #20566

Masanetz, 2020-12-25:
Looks great, menu is working...

Many thanks!

PR #20566 (2020-12-28): Fix Menu Mixer for Color UI
### Description

This fix Menu Mixer for Color UI.

It add some glue code for Color UI works with `MENU_ITEM_ADDON`, so more screens could benefit from this. I added a task for my self to check all `MENU_ITEM_ADDON` later.

### Benefits

Fix #20023
closed 2020-12-28 by rhapsodyv