MarlinFirmware/Marlin #20024: [BUG] Cannot enable speaker for FYSETC S6 (v1.2) board

mak0t0san, 2020-11-04:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[FYSETC S6 Marlin Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5489093/FYSETC.S6.Marlin.Configs.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Uncomment `#DEFINE SPEAKER` in configuration.h
2. Build firmware
3. See failure

**Expected behavior:**

Build succeeds, able to hear tones on speaker

**Actual behavior:**

Build fails
```
Marlin\src\HAL\STM32\timers.cpp:278:41: error: static assertion failed: One or more timer conflict detected
  278 | static_assert(verify_no_duplicate_timers(), "One or more timer conflict detected");
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~~^~
*** [.pio\build\FYSETC_S6\src\src\HAL\STM32\timers.cpp.o] Error 1
```

#### Additional Information

Proposed resolution:
In pins_FYSETC_S6.h, add a line to specify a different timer (8 for example) for the STEP_TIMER:
```
#define STEP_TIMER              8
```

mak0t0san, 2020-11-04:
@GerogeFu , would you be able to take a look at this please?

sjasonsmith, 2020-11-06:
@mak0t0san I have posted a pull request to fix this. I hadn't even noticed that you already solved the problem yourself :)

You are free to post pull requests to fix issues you find. You do not have to wait for Geroge to make the change.

mak0t0san, 2020-11-06:
> @mak0t0san I have posted a pull request to fix this. I hadn't even noticed that you already solved the problem yourself :)
> 
> You are free to post pull requests to fix issues you find. You do not have to wait for Geroge to make the change.

That was going to be my next step. I wanted to make sure that this was a real issue, and see if there's a preference on which timer to use.

GerogeFu, 2020-11-07:
Acturally , i prefer to change S6 variant #define TIMER_TONE              TIM6

sjasonsmith, 2020-11-07:
I changed the stepper timer because it already is known to work with TIM10 on the Rumba32. I don’t believe it causes any conflicts.

GerogeFu, 2020-11-07:
Ok , i will test it too , thanks for this  PR for S6

sjasonsmith, 2020-11-07:
@GerogeFu I have a general question for you. Do you think it would be easier to remover the default TEMP/STEP timers, and require each board to specify timers in the pins file?

GerogeFu, 2020-11-07:
I don't think so. If people really want to change the timers , then they can specify timers in pins file. But i recommend Marlin to check the TIMER variant of related boards when need to change the default timer.

sjasonsmith, 2020-11-07:
@mak0t0san my PR has been merged. Can you verify that this works, and close the bug if it seems resolved?

sjasonsmith, 2020-11-07:
@mak0t0san I know it has not been long since the change merged, but I'm going to go ahead and close this issue. Please reply if you still encounter problems, and we can re-open if necessary.

closed 2020-11-07 by sjasonsmith

mak0t0san, 2020-11-09:
> @mak0t0san my PR has been merged. Can you verify that this works, and close the bug if it seems resolved?

Verified that `STEP_TIMER 10` works.  Build succeeds and speaker tones work as expected.

sjasonsmith, 2020-11-09:
Just a comment for anybody that reads this and is familiar with STM32 timers. Some timers (including timer 10) run in a different source clock rate than other timers. Previously this had to be overridden for a specific board, but this is no longer necessary. About a week ago I changed the HAL so that it would read the timer’s clock rate from the ST framework, so any timer should be able to be used without clock rate concern.