MarlinFirmware/Marlin #20026: CREALITY v4.2.7 - Error writing to EEPROM! // Error:Field esteppers mismatch

misiakS, 2020-11-04:
### Bug Description

After flashing bugfix-2.0.x to new Creality v4.2.7 board on Ender 5 with BLtouch I cannot write to EEPROM. Stored settings are lost after reboot.
Using either Pronterface terminal after M500 or printer LCD "Save settings" I get following log:

echo:Error writing to EEPROM!
Error:Field esteppers mismatch.
[ERROR] Error:Field esteppers mismatch.
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
... and so on many times...

Without defining DEBUG_EEPROM_READWRITE I get: Error:EEPROM datasize error.

I used hints from threads:  #10263, #12048, #11757, #11883, #12015, #11699. These were:

- different SD cards
- M502 > M500 > M500
- EEPROM_OFFSET set to 100
- initialize EEPROM
- active DEBUG_EEPROM_READWRITE in Configuration.h
- commented items according to [post](https://github.com/MarlinFirmware/Marlin/issues/11883#issuecomment-423242712

What's important in terms of EEPROM functionality - Creality firmware, dedicated for this board works. Settings are stored.
 
I was trying the solution with discrepancy of NUM_SERVOS vs NUM_SERVO_PLUGS over code, but this was already patched a year(s) ago. But I see now there is  EEPROM_NUM_SERVOS along NUM_SERVO. Could be the same issue?

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5490328/Configs.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. compile attached config files in bugfix-2.0.x
2. flash it to board though SD card (mine is Creality v4.2.7)
3. customize settings & store them
4. reboot and check stored settings

**Expected behavior:**

Stored settings remain stored after reboot.

**Actual behavior:**

Stored settings are replaced by Hardcoded defaults after reboot.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

legend069, 2020-11-05:
#define EEPROM_AUTO_INIT  // Init EEPROM automatically on any errors.
 this will load defaults if there are eeprom errors
have you changed this value ?
also it looks like you have the wrong drivers defined in firmware
#define X_DRIVER_TYPE  TMC2208_STANDALONE
#define Y_DRIVER_TYPE  TMC2208_STANDALONE
#define Z_DRIVER_TYPE  TMC2208_STANDALONE
when the actual board drivers are
TMC2225 
https://www.amazon.com.au/Creality-Upgraded-Motherboard-Mainboard-TMC2225/dp/B084ZDX7NY

victor7376, 2020-11-05:
> #define EEPROM_AUTO_INIT // Init EEPROM automatically on any errors.
> this will load defaults if there are eeprom errors
> have you changed this value ?
> also it looks like you have the wrong drivers [...]

There is a topic regarding these drivers and the 2208 are compatible with 2225.  I use the same drivers with the same board on the Ender 3.

Topic:
https://github.com/MarlinFirmware/Marlin/issues/15632

misiakS, 2020-11-05:
> TMC2225

Yes, I uncommented EEPROM_AUTO_INIT to skip error at bootup. Same result without it.
Well I missed the actual driver mismatch. But as @victor7376 said, the firmware seems to be same for 2208 and 2225. Here are details https://wiki.fysetc.com/TMC2225/

legend069, 2020-11-05:
X, Y, Z, E0 [, E1[, E2...]]
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 78, 78, 787, 93 }
is that a typo, or do you want 787 steps per mm on the z motor?

sjasonsmith, 2020-11-06:
This is failing when trying to communicate with the I2C flash on the board and that is not working. It works fine when using the Ender 3 example for this board. I haven't spent the time yet to figure out what is different between them.

thisiskeithb, 2020-11-06:
> There is a topic regarding these drivers and the 2208 are compatible with 2225. I use the same drivers with the same board on the Ender 3.

There's also a comment in `Configuration.h` explaining what to do when you have TMC2225s or TMC2226s:
https://github.com/MarlinFirmware/Marlin/blob/c574bcce880e769ae3851304da2a1b1ee19466ed/Marlin/Configuration.h#L677

legend069, 2020-11-06:
> TMC2225

that comment line isn't there in my config.h or OP
and nothing comes up when i search "TMC2225"

thisiskeithb, 2020-11-06:
> that comment line isn't there in my config.h or OP

Update to the latest `bugfix-2.0.x`. It was added a couple days ago.

sjasonsmith, 2020-11-06:
Please let this issue remain focused on the EEPROM concern.

sjasonsmith, 2020-11-06:
The EEPROM failure is caused by the following line. Comment it out and your EEPROM should work:

`#define SERIAL_PORT_2 -1`

The `-1` serial port is for the STM32's built-in USB connection, which this board does not use. It uses a, external serial-USB converter chip. Commenting this line out removes no functionality from your hardware.

misiakS, 2020-11-06:
> The EEPROM failure is caused by the following line. Comment it out and your EEPROM should work:
>
> `#define SERIAL_PORT_2 -1`
>
> The `-1` serial port is for the STM32's built-in USB connection, which [...]

@sjasonsmith This is fix. You are my hero. I was idealess after 3 full days of debugging (novice). Let me know if I can help you anytime.
Damn, I made this bug myself. 

Thank you all guys for help, I tried your steps. Appreciate!

Regards,
Michal

closed 2020-11-06 by michalsovcik

sjasonsmith, 2020-11-06:
I am glad it is working. I am preparing a pull request which should detect this problem and make it easier to find for other users.

I suspect that the I2C and USB share pins. Enabling the USB serial interferes with the I2C pins.

michalsovcik, 2020-11-06:
> I am glad it is working. I am preparing a pull request which should detect this problem and make it easier to find for other users.
> 
> I suspect that the I2C and USB share pins. Enabling the USB serial interferes with the I2C pins.

Yes, correct as you are saying. USB shar pin with other buses. [Datasheet ](https://www.mouser.sk/datasheet/2/389/stm32f103rc-1851170.pdf) p.35:
[image]