MarlinFirmware/Marlin #20027: [BUG] Bilinear subdivision unwanted Z movement

jiridoko, 2020-11-04:
### Bug Description

Exactly the same thing as this bug describes https://github.com/MarlinFirmware/Marlin/issues/14963 here. It is still happening and this bug has not been resolved. Using LPC1769 mcu (skr 1.4 turbo) it is filling one corner with null Z offset data and the prints in this area are always in zero Z height whatever Z height the print is trying to be at. It is squishing all the layers.

### Configuration Files

https://github.com/jiridoko/Marlin/blob/mk1_4/Marlin/Configuration.h
https://github.com/jiridoko/Marlin/blob/mk1_4/Marlin/Configuration_adv.h

### Steps to Reproduce

1. Call G28
2. Call G29

**Expected behavior:**

the offset table should have an actual value in all columns and rows of the table

**Actual behavior:**

```
echo:Home XYZ First
ok
X:114.00 Y:148.00 Z:10.00 E:0.00 Count X:9120 Y:11840 Z:4000
ok
Bilinear Leveling Grid:
      0      1      2      3      4
 0 +0.540 +0.242 -0.035 -0.268 -0.655
 1 +0.385 +0.127 -0.095 -0.378 -0.885
 2 +0.275 +0.082 -0.075 -0.213 -0.555
 3 +0.250 +0.117 -0.045 -0.208 -0.560
 4 +0.425 +0.227 +0.130 +0.087 -0.195

Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9       10       11       12
 0 +0.54000 +0.44009 +0.34018 +0.24250 +0.14685 +0.05343 -0.03500 -0.11009 -0.18019 -0.26750 -0.38519 -0.52009  =======
 1 +0.48667 +0.39069 +0.29471 +0.20157 +0.11305 +0.02737 -0.05796 -0.13693 -0.21555 -0.31435 -0.44691 -0.59966  =======
 2 +0.43333 +0.34129 +0.24924 +0.16065 +0.07925 +0.00131 -0.08093 -0.16377 -0.25091 -0.36120 -0.50864 -0.67923  =======
 3 +0.38500 +0.29787 +0.21074 +0.12750 +0.05296 -0.01769 -0.09500 -0.17639 -0.26444 -0.37750 -0.53000 -0.70750  =======
 4 +0.34185 +0.26126 +0.18067 +0.10435 +0.03678 -0.02652 -0.09463 -0.16387 -0.23792 -0.33694 -0.47600 -0.64004  =======
 5 +0.30370 +0.23063 +0.15756 +0.08898 +0.02810 -0.02830 -0.08537 -0.13712 -0.18955 -0.26583 -0.38163 -0.52128  =======
 6 +0.27500 +0.20954 +0.14407 +0.08250 +0.02667 -0.02528 -0.07500 -0.11472 -0.15222 -0.21250 -0.31148 -0.43324  =======
 7 +0.25296 +0.19672 +0.14049 +0.08546 +0.03180 -0.02066 -0.07111 -0.11134 -0.14957 -0.20972 -0.30706 -0.42631  =======
 8 +0.24037 +0.19344 +0.14651 +0.09731 +0.04417 -0.01124 -0.06611 -0.11230 -0.15796 -0.22472 -0.32685 -0.45009  =======
 9 +0.25000 +0.20694 +0.16389 +0.11750 +0.06556 +0.01028 -0.04500 -0.09213 -0.13926 -0.20750 -0.31093 -0.43546 -0.56000
10 +0.29352 +0.24526 +0.19700 +0.14861 +0.09955 +0.05036 +0.00259 -0.03529 -0.07176 -0.13065 -0.22739 -0.34657 -0.46574
11 +0.35926 +0.30036 +0.24146 +0.18806 +0.14255 +0.10254 +0.06630 +0.04268 +0.02283 -0.02157 -0.10814 -0.21926 -0.33037
12 +0.42500 +0.35546 +0.28593 +0.22750 +0.18556 +0.15472 +0.13000 +0.12065 +0.11741 +0.08750 +0.01111 -0.09194 -0.19500

X:207.00 Y:225.00 Z:7.90 E:0.00 Count X:16560 Y:18000 Z:3082
ok
```

#### Additional Information

* https://youtu.be/s-VIow97JhI

daleckystepan, 2020-11-04:
I am also experiencing this bug. You can see my configs:

https://github.com/daleckystepan/Marlin/blob/geeetech_a10/Marlin/Configuration.h
https://github.com/daleckystepan/Marlin/blob/geeetech_a10/Marlin/Configuration_adv.h

Config highlights:
SKR 1.4 Turbo, FIX_MOUNTED_PROBE (Mini differential IR), GRID_MAX_POINTS_X 3, ABL_BILINEAR_SUBDIVISION, BILINEAR_SUBDIVISIONS 3.

bellerofonte, 2020-11-06:
Hi!
I'm experiencing the same bug since at least 2.0.6
Decided to disable subdivisions and increasing MAX_POINTS

sjasonsmith, 2020-11-16:
I posted a PR which should fix this: #20143

I am going to close this as a duplicate, even though it is slightly older than the other issue. I didn't realize this was already here when I investigated the other.

sjasonsmith, 2020-11-16:
Duplicate of #20122

closed 2020-11-16 by sjasonsmith