MarlinFirmware/Marlin #20028: FAST_PWM_FAN causes BTT002 part cooling fan to run at 100%

thisiskeithb, 2020-11-04:
### Bug Description

https://github.com/MarlinFirmware/Marlin/pull/20025 allows boards in the STM32 HAL to use `FAST_PWM_FAN`, but enabling it on a BTT002 with the latest `bugfix-2.0.x` (a1cce36), the part cooling fan runs at 100%.

Just to make sure it wasn't a stuck mosfet, I re-flashed with `FAN_SOFT_PWM` enabled and the part cooling fan works as it should.

### Configuration Files

Use Marlin/configs from https://github.com/thisiskeithb/Marlin/tree/myconfig/MK3S-BTT002-TMC2209 and enable `FAST_PWM_FAN`/disable `FAN_SOFT_PWM`. This is based on the latest `bugfix-2.0.x` (a1cce36).

### Steps to Reproduce

1. Compile Marlin/Configs from https://github.com/thisiskeithb/Marlin/tree/myconfig/MK3S-BTT002-TMC2209 and enable `FAST_PWM_FAN`/disable `FAN_SOFT_PWM`.
2. Observe part cooling fan running at 100% despite it showing as 0%/off.

**Expected behavior:**

Part cooling fan speed to be controllable.

**Actual behavior:**

Part cooling fan runs at 100% even though it's set to 0%.

#### Additional Information

I'll try to test some other STM32F4-based boards like the SKR Pro & GTR later to see if those are also affected.

thisiskeithb, 2020-11-05:
I'm working with @sjasonsmith via discord since there's some issues with default fan frequencies on these STM32F4-based boards as well.

sjasonsmith, 2020-11-05:
It looks like it works on the SKR Pro at least, but the default rate is using a formula designed for AVR, resulting in a default rate of 329.4 kHz.

It is unclear at this point whether it is also working on the BTT002. I suspect one of the following right now:
1. BTT002 never worked with "normal" PWM, as Keith has always been using Soft PWM with it. This might be due to a timer conflict.
2. It is actually working, but the high frequency doesn't play nicely with the fans in use

sjasonsmith, 2020-11-05:
It seems the PWM assignments in the BTT002 was copied from the SKR Pro and never updated. We’ll have to update that and then I expect it will work.

Minims, 2020-11-05:
With this option my Radial FAN on FAN0 is 0 or 100% even if I'm a 20%. I'm on SKR PRO 1.1

sjasonsmith, 2020-11-05:
@Minims I validated the feature works with an oscilloscope on all three fan channels on an SKR Pro. Fan PWM is tricky. Some fans prefer high frequencies, others prefer low. Some will spin at 100% even when the scope shows a 1% duty cycle.

There are electrical behaviors occurring which are beyond my understanding. Suffice it today that the “best” option here is not universal and needs to be tailored for your hardware.

Minims, 2020-11-05:
Ok :-), I'm back to FAN_SOFT_PWM and it works great. I would have like test this new feature, but I might need to try an other fan. Thx

thisiskeithb, 2020-11-05:
> I'm back to FAN_SOFT_PWM and it works great.

That's where I'll be after Jason's BTT002 PeripheralPins.c fix. I don't really have many fan options on the BTT002 since it's for the Prusa & uses 5v fans.

`FAST_PWM_FAN` is a nice feature, but seems a bit finicky on STM32. At least for now.

closed 2020-11-05 by thisiskeithb