MarlinFirmware/Marlin #20029: Junction Deviation still rearing it's ugly stuttering head.

thierryzoller, 2020-11-04:
**Version**: Latest Bugfix
**Duplicate/Related :** https://github.com/MarlinFirmware/Marlin/issues/17342

### Bug Description

Having Junction Deviation enabled leads to stuttering XY Diagonal movements. See Videos below.
Disabling JD solves the issue. 

### Configuration Files
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5490782/Marlin.zip)

TMC2209 UART Mode
Hybrid Threshold (doesn't change the effect)

#### Additional Information
* Video 1 : https://youtu.be/kngqQIPUP1c
_JD Enabled (See Diagonal Movement)

* Video 2 : https://youtu.be/yOadfORpKuw
_Classic Jerk Enabled (See - TMC 2209 Hybrid Threshold straight but noisy movement)_

thisiskeithb, 2020-11-04:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/17342

closed 2020-11-04 by thisiskeithb

thierryzoller, 2020-11-04:
17342 is about a subset of calculations that I think do not apply here. So why close it? Please leave someone chime in on whether or not it is really a duplicate. Root cause may be different. I don't see how you would close this.

sjasonsmith, 2020-12-20:
@thierryzoller are you still having issues with this?

thierryzoller, 2020-12-21:
I disabled JD since and do not see the issue. I can try recompilig it with JD if there is reason to believe it should not reproduce

sjasonsmith, 2020-12-21:
I don’t have any reason to believe it has improved, but your configs were newer than that bug this duplicates, so could be a better starting point for anyone wanting to debug it.

I’m hoping we’ll knock out quite a few bugs over the next couple weeks, but I can’t really promise exactly which ones we can get to.