MarlinFirmware/Marlin #20031: [FR] Add support for Qidi thermocouples

awesomeguy17123, 2020-11-05:
### Description

My QIDI X-one2's thermocouple is not supported. It would be cool to see it be supported.

ellensp, 2020-11-05:
How do you know its not supported, what is it?
"QIDI X-one2" is a machine brand

awesomeguy17123, 2020-11-05:
I have tried every thermistor/thermocouple. none have worked

ellensp, 2020-11-05:
According to a e-bay listing the controller is this https://i.ebayimg.com/images/g/Fv0AAOSweP9eTxlD/s-l1600.jpg  but cant find any details, like how it talks to a thermocouple

awesomeguy17123, 2020-11-05:
Yes. so for the board, I selected chitu3d_v5 and it seems to work. All but the thermocouple works just fine, even the heated bed thermistor works set to 1.

awesomeguy17123, 2020-11-05:
also, this might be a chitu V3.9. Here is the manual with all the specs of it.
[Chitu V3.9 users manual.pdf](https://github.com/MarlinFirmware/Marlin/files/5496179/Chitu.V3.9.users.manual.pdf)

Grogyan, 2020-11-05:
Temperature Sensor Interface: 3 paths of 100K NTC (thermistor)
2 paths of MAX6675（thermocouple）

from that manual.

You need to ensure that SD is enabled, and choose which SPI port the MAX6675 is tied to,
If for the onboard storage, then in config_adv, you need to set onboard for the SD card.

awesomeguy17123, 2020-11-05:
How would I find what SPI port it is tied to. I have just enabled SD Support.

Grogyan, 2020-11-05:
@awesomeguy17123 Use a multimeter to probe the max6675 MISO, SCK and CS pins, to the micro.
The datasheet for which will tell you which pin is what.

awesomeguy17123, 2020-11-05:
Ok i'll try that real quick

awesomeguy17123, 2020-11-05:
I'm not getting any readings other than 0. Does it have to be powered on or am I touching the wrong pins?

awesomeguy17123, 2020-11-05:
[photos.zip](https://github.com/MarlinFirmware/Marlin/files/5497397/photos.zip)
Here are pictures of the mainboard. Hope this helps in some way.

ellensp, 2020-11-06:
Cant open your file photoes.zip. Zip archive seems truncated.

awesomeguy17123, 2020-11-06:
[Photos(Working).zip](https://github.com/MarlinFirmware/Marlin/files/5501595/Photos.Working.zip)
sorry about that. I must have stopped mid-zip.

clinde04, 2020-11-12:
I've decided to purchase a new mainboard and some other supplies. Thank you for all your help though.

awesomeguy17123, 2020-11-12:
I've decided to purchase a new mainboard and some other supplies. Thank you for all the help though.

closed 2020-11-12 by awesomeguy17123