MarlinFirmware/Marlin #20037: [BUG & FIX] Sensorless homing/probing sensitivity on Delta printers

Sewyboowy, 2020-11-05:
When homing, I am using a lower current than usual such that X_CURRENT_HOME = X_CURRENT/2. After dialling in the sensitivity, I found that the Z stepper sensitivity was significantly lower (ie didn't stall as easily). I changed tried changing the steppers and the steppers, and the behaviour was always on the z stepper. I have now fixed this issue by editing the G28.cpp file which is attached with the original. Please forgive me if what i've done is completely wrong, I only have experience in python! I also think my fix would cause issues with cartesian printers but that could probably be fixed with a #if ENABLED(DELTA).

I am also using sensorless probing, and am finding that the z stepper doesn't stall at all (or at least skips steps before stallguard detects it), although the feature still works as the x and y steppers stall correctly. This results in the effector tilting significantly before during probing. I have not yet fixed this issue but would definitely appreciate help with that. I'd imagine its a similar issue. I started editing probe.cpp but haven't got a fix yet, it's attached anyway with the original. Config files also attached.

### Steps to Reproduce

On Delta printer, setup sensorless homing and dial in sense settings with homing current less that typical current. On homing, Z stepper will not stall as easily as X and Y.
Try sensorless probing, Z axis will not respond to stalls.
[Sensorless homing and probing.zip](https://github.com/MarlinFirmware/Marlin/files/5496158/Sensorless.homing.and.probing.zip)

boelle, 2020-11-05:
its a know fact that sensorless homing does not work on screw driven systems. to much torque no matter what setting

Sewyboowy, 2020-11-05:
Sorry if it wasn't clear, this is for a delta printer, so no screw driven system. So in a delta we still have an X, Y and Z tower/stepper, and what I think is happening is Marlin is generally written for cartesian printers, so the code for X and Y isn't the same as Z. However in the delta, X Y and Z all have to work the same.

Sewyboowy, 2020-11-05:
If you look at what i changed in G28.cpp, all I did was copy code already originally written only for X and Y, and pasted it again for Z.

sjasonsmith, 2020-11-06:
If you already a fix for this please post a pull request with the changes. It takes a lot of effort to extract changes from a zip file, and nobody looking at them will have the ability to test them as you do.

In the past there have been issues with delta and sensorless homing, and I think the users abandoned the feature. If you have it working you may be the de-facto expert at this point.

Sewyboowy, 2020-11-06:
Ok, I only have a fix for the homing issue at present. Yeah I've had lots of issues in the past and contributed to other threads. Sensorless homing works ok with stock marlin, and I think without issue. 

I'll try tidy up the fix so that it doesn't interfere with other printer types. I don't have another printer though, so it'd be great if someone could test it on a cartesian machine to see if it still works!

sjasonsmith, 2020-11-06:
You can always post it as a draft pull request and point out any concerns you have.

Sewyboowy, 2020-11-06:
I've managed to get probing to be triggered by the z stepper stalling as well as x and y, all i did was uncomment #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN in config, not sure why I didn't have that before. I don't think they're running at the reduced current though.

Foxies-CSTL, 2020-11-07:
> However in the delta, X Y and Z all have to work the same.

Very interesting your post, I am currently testing on a QQS-Pro TMC2209 in UART mode. I have not yet activated "StallGuard2" to remove endstops. I can help you with physical tests on my printer but the coding is too complicated for me ;-). 
if I can help you ......

alexandremstf, 2020-11-08:
I have a similar problem on corexy machine with belt drive system on z axis. I can use sensorless homing on x and y but on z axis the sensibility is really strange. In my case i just want to use z sensorless homing to identify the max homing and after use a mesh bed leveling. But until now i couldnt, someone has some tip?

sjasonsmith, 2020-11-08:
@alexandremstf its probably best to keep this issue focused on delta. Delta homing is unique since it homes all axis simultaneously.

Sewyboowy, 2020-11-08:
> I have a similar problem on corexy machine with belt drive system on z axis. I can use sensorless homing on x and y but on z axis the sensibility is really strange. In my case i just want to use z [...]

That sounds like a similar issue and I think what I'm doing would likely solve you're problem. You'll have to be careful as it's not recommended to use sensorless homing/probing with lead screw systems as they generate too much torque. You can try it by swapping the G28.cpp file I posted for your current one. Make sure your keep a copy of your current g28.cpp tho.

I must emphasize that I really don't know what I'm doing, so try at your own risk. Sensorless probing is not fixed yet so don't try with a lead screw system!

sowd0726, 2020-11-09:
@Sewyboowy 
Let me ask a question in your modification on probe.cpp.
Your change is to enable same behavior on Z as XY is doing in DELTA case . Right ?
But it looks your code change in probe.cpp actually do nothing in DELTA case.
Would you elaborate the reason why this change is required in DELTA ?

Sewyboowy, 2020-11-09:
> @Sewyboowy
> Let me ask a question in your modification on probe.cpp.
> Your change is to enable same behavior on Z as XY is doing in DELTA case . Right ?
> But it looks your code change in probe.cpp actually do nothing in DELTA case.
> Would you elaborate the reason why this change is required in DELTA ?

Yes you're right. I'm working on the probe.cpp now and have basically added the current code from g28.cpp into probe.cpp. So far i've got it working so that the current reduces whilst moving to the first probing position and during the first probe, but after that the current returns to the original value.

This is where my lack of understanding of C++ rears it's ugly head! Many hours later I'm still struggling to figure out why this happens. My plan now is to update to the latest marlin and start from scratch on my editing of probe.cpp, as i've spent so much time editing the original that its getting very messy and hard to look at haha

sowd0726, 2020-11-10:
Sorry for many questions. Let me ask a more question.
Which version are you working for ?
The reason why I am asking is ...

I am checking bugfix_2.0.x and it looks G28.cpp still needs to be changed to define the homing current for Z axis.
On the other hand , when I tried 2.05/2.06 # I am not sure which # it has something wrong with sensor less homing for Z-axis in Delta.
But 2.07 sensor less homing using 2209 is working fine with Delta in my hand as far as not defining homing current for Z axis.
It looks there was some fixing on sensor less homing with Delta recently.
Since your ZIP including file based on elder version.
Then I am afraid the case when your are currently working on not latest version.

Sewyboowy, 2020-11-10:
No worries,

Yeah I was using 2.05 before I think, started working on switching to the latest 2.0.7.2 and the endstops for probing seems to have changed as my z_probe and y_min enstops (although not x_min or z_min) are "TRIGGERED" even with pulldown enabled. This is all using the exact same settings as I used on my 2.0.5 build.

I haven't had much time to figure this out yet though but hopefully i'll get it working soon.

Franzt151, 2020-11-12:
I'm having an issue getting my Z axis to home with my delta printer. I'm using an SKR 1.4T with TMC2209 drivers on 2.0.7. I tried looking at the G28 file and did not see where to add the Z-axis. Is there something you guys recommend to try?

Sewyboowy, 2020-11-12:
Yeah since moving to 2.0.7 I'm having issues too. It's a good idea to check the endstop are configured correctly with m119. You want to make sure that all endstops are open by default. It can be a bit difficult to troubleshoot sensorless homing as they only trigger for a short period of time. You can test if the endstop is triggering correctly by connecting the endstop to 3.3V or GND and the endstop state should change between triggered and open. At some point in the past I've had it where the z axis was triggered by default stopping it from homing.

Franzt151, 2020-11-13:
@Sewyboowy Thank you for that. When I run m119 z-axis is open. When I connect the Z end stop to 3.3v it triggers. I then ran a G28 and manually triggered the end stop and it triggers and homed successfully. Is there something that would cause the driver to not trigger the DIAG pin?

Franzt151, 2020-11-13:
I got it working with 2.0.7. I'm not quite sure what I did to make it work though. I hooked up my oscilloscope to see what the DIAG pins were doing. They triggered correctly and successfully homed. I did not make any changes to my G28 file but here are my config files if you want to look.
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5535018/Config.zip)

Sewyboowy, 2020-11-13:
@Franzt151 Thanks!

Sewyboowy, 2020-11-15:
With the help of your config files I've gotten sensorless homing working on 2.0.7 but not probing. If I enable x, y and z min plugs for sensorless probing, I get an issue where y_min and z_min are always triggered with m119. x_min is still open though. I checked the pins.h file for my board (SKR PRO V1.1) and min and max endstops are the same pin for delta configs, so x, y and z max/min endstops should always read the same. Pretty sure that's just a software bug.

Furthermore, when I manually connect the endstop pins to 3.3v or GND, nothing changes with m119. Even weirder, sensorless homing still works (sort of), the x axis homes fine, but y and z don't bump correctly. The y and z axes don't move down for the bump (presumably because the y and z min endstops are triggered), but then try to move up again which causes the steppers to skip and/or belt slip.

Sewyboowy, 2020-11-15:
I wondered if this may be fixed if I set the x y and z min software pins to their own physical pins. Will try when I have time.

Sewyboowy, 2020-11-15:
> I wondered if this may be fixed if I set the x y and z min software pins to their own physical pins. Will try when I have time.

It worked! Finally I can work on the original issue!

Sewyboowy, 2020-11-15:
So the fix in G28 is only for the homing current. If the z axis isn't homing at all it's probably cus your endstops aren't set correctly. As I said above, this can be caused if you have x,y,z min enabled I think because of a bug. Use m119 to check your endstops before homing

closed 2020-12-23 by github-actions[bot]

StarSOziOne, 2020-12-31:
Greetings my friend, thank you for your hard work. I don't know much about coding (just copy paste and simple commands)
So the question I want to ask is, I wonder if I replace the "G28.cpp and probe.cpp" file in the "Sensorless.homing.and.probing" file you shared in your first message, with the files in the marlin 2.0.x I am currently using. Do I have any problem in the compilation?
Also, do you have changes in "Configuration.h and Configuration_adv.h" except for the standard delta settings, do I have to replace them (of course by configuring them according to my own delta printer) The drivers I use are TMC5161

StarSOziOne, 2020-12-31:
my marlin version is bugfix-2.0.x g28 and when i switch probe files i get a few different errors. I will try now in version 2.0.7.2